flipperdevices/flipperzero-firmware #3461: Python path is incorrect during install

jpitz31, 2024-02-18:
### Describe the bug.

during install of fbt vscode_dist,  my virus protection stopped the install process.  I stopped my virus protection and attempted to run fbt vscode_dist  --clean 

see below error:

If I delete the flipperzero-firmware folder and re git clone --recursive, and then try to run fbt vscode_dist I get the same error.

I now cannot install the tool chain using fbt vscode_dist
(see below)

(there is no toolchain folder under the flipperzero-firmware folder.)
searching for this error it recommends to unset PYTHONHOME env variable,  it is not set in my environment,  but is being set by the vscode_dist process.

(fbt) C:\FlipperFW\flipperzero-firmware>fbt vscode_dist
Python path configuration:
  PYTHONHOME = 'C:\FlipperFW\flipperzero-firmware\toolchain\x86_64-windows\python'
  PYTHONPATH = (not set)
  program name = 'python'
  isolated = 0
  environment = 1
  user site = 0
  safe_path = 0
  import site = 1
  is in build tree = 0
  stdlib dir = 'C:\FlipperFW\flipperzero-firmware\toolchain\x86_64-windows\python\Lib'
  sys._base_executable = 'C:\\Users\\Joe Pitz\\AppData\\Local\\Programs\\Python\\Python312\\python.exe'
  sys.base_prefix = 'C:\\FlipperFW\\flipperzero-firmware\\toolchain\\x86_64-windows\\python'
  sys.base_exec_prefix = 'C:\\FlipperFW\\flipperzero-firmware\\toolchain\\x86_64-windows\\python'
  sys.platlibdir = 'DLLs'
  sys.executable = 'C:\\Users\\Joe Pitz\\AppData\\Local\\Programs\\Python\\Python312\\python.exe'
  sys.prefix = 'C:\\FlipperFW\\flipperzero-firmware\\toolchain\\x86_64-windows\\python'
  sys.exec_prefix = 'C:\\FlipperFW\\flipperzero-firmware\\toolchain\\x86_64-windows\\python'
  sys.path = [
    'C:\\Users\\Joe Pitz\\AppData\\Local\\Programs\\Python\\Python312\\python312.zip',
    'C:\\FlipperFW\\flipperzero-firmware\\toolchain\\x86_64-windows\\python\\DLLs',
    'C:\\FlipperFW\\flipperzero-firmware\\toolchain\\x86_64-windows\\python\\Lib',
    'C:\\Users\\Joe Pitz\\AppData\\Local\\Programs\\Python\\Python312',
  ]
Fatal Python error: init_fs_encoding: failed to get the Python codec of the filesystem encoding
Python runtime state: core initialized
ModuleNotFoundError: No module named 'encodings'

Current thread 0x000032e0 (most recent call first):
  <no Python frame>

(fbt) C:\FlipperFW\flipperzero-firmware>

### Reproduction

Using avast virus protection.

run the install fbt vscode_dist

virus protection stops the install of libwinpthread-1.dll

delete flipperzero-firmware and re git clone -- recursive

attempt to run fbt vscode_dist --clean

throws the above error

### Logs

```Text
Windows 11
```

### Anything else?

let me know if you need anything else.

hedger, 2024-02-19:
When dealing with borked toolchain configuration, just remove "toolchain" directory completely.

https://www.virustotal.com/gui/file/31c54186ac2aaa41a420fa78f7ff97a34e8d71e3e1dafe20b274e3ade4c1575a?nocache=1 - this is Virustotal result (0/71) for libwinpthread-1.dll from toolchain v32, in branch https://github.com/flipperdevices/flipperzero-firmware/pull/3451. 

Could you please check with that branch and confirm the exact offending file?

jpitz31, 2024-02-19:
Hey hedger, toolchain folder was not there in the first place.
I have removed entire firmware and re-cloned, error still exits.
Let me know how to check for the offending file?

Thanks

Joe

-- 
http://joestechblog.com

*//
--------------------------------------------------------------------------------------*
*Old C programmers never die, they just get a new function pointer.*

*..."It began upon the following occasion: It is allowed on all hands, that
the primitive way of breaking eggs before we eat them, was upon the larger
end: but his present Majesty's grandfather, while he was a boy, going to
eat an egg, and breaking it according to the ancient practice, happened to
cut one of his fingers. Whereupon the Emperor his father published an
edict, commanding all his subjects, upon great penalties, to break the
smaller end of their eggs."  Jonathan Swift*

hedger, 2024-02-19:
@jpitz31 please try running `fbt` from powershell. Running it in CMD contaminates execution environment after failed environment setup or toolchain installation, breaking commands after it.

Also provide full log from running `fbt` after deleting `toolchain` folder.

jpitz31, 2024-02-21:
Hedger, removed the firmware folder and re-cloned -- recursive
I then loaded a power shell and ran ./fbt
Upon completion I received the following error:
Mode                 LastWriteTime         Length Name
----                 -------------         ------ ----
d-----         2/20/2024   7:33 PM                toolchain
Extracting Windows toolchain..moving..linking to 'current'..Junction created for C:\FlipperFW\flipperzero-firmware\toolchain\x86_64-windows\..\current <<===>> C:\FlipperFW\flipperzero-firmware\toolchain\x86_64-windows
done!
Cleaning up temporary files..done!

scons: *** Toolchain version is not supported. Allowed: (' 12.3.', ' 13.2.'), toolchain:    Stop.

At this point I cannot run ./fbt as I get the same error as show above.

No log files were created.

Thanks
Joe

hedger, 2024-03-01:
@jpitz31 we've merged multiple fixes and improvements for the toolchain, could you please check out latest `dev` and see if your issue was resolved?

jpitz31, 2024-03-06:
Hey Hedger,  I tried the new build,  several times,

The ./fbt command is stopping after the following line:
C:\flipper\flipperzero-firmware>fbt
        CPP     lib\app-scened-template\view_modules\byte_input_vm.cpp
        CPP     lib\app-scened-template\view_modules\dialog_ex_vm.cpp
        CPP     lib\app-scened-template\view_modules\popup_vm.cpp
        CPP     lib\app-scened-template\view_modules\submenu_vm.cpp
        CPP     lib\app-scened-template\view_modules\text_input_vm.cpp
        CPP     lib\app-scened-template\text_store.cpp
        ICONS   build\f7-firmware-D\assets\compiled\assets_icons.c
        PROTO   assets\protobuf\application.proto
        DOLPHIN blocking
        DOLPHIN internal
        CC      lib\bit_lib\bit_lib.c
        CC      lib\ble_profile\extra_profiles\hid_profile.c
2024-03-06 18:07:22,429 [INFO] Processing Dolphin sources
2024-03-06 18:07:22,430 [INFO] Loading data
2024-03-06 18:07:22,430 [INFO] Loading directory assets\dolphin\blocking
2024-03-06 18:07:22,431 [INFO] Processing Dolphin sources
2024-03-06 18:07:22,431 [INFO] Loading data
2024-03-06 18:07:22,432 [INFO] Loading directory assets\dolphin\internal
2024-03-06 18:07:22,433 [INFO] Packing
2024-03-06 18:07:22,434 [INFO] Packing
2024-03-06 18:07:22,530 [INFO] Complete
2024-03-06 18:07:22,533 [INFO] Complete
        CC      build\f7-firmware-D\assets\compiled\assets_dolphin_internal.c
        CC      build\f7-firmware-D\assets\compiled\assets_dolphin_blocking.c
        CC      build\f7-firmware-D\assets\compiled\assets_icons.c

Thanks

hedger, 2024-03-21:
It should work now, please re-open if you experience any issues with Python with code in current `dev` branch.

closed 2024-03-21 by hedger