flipperdevices/flipperzero-firmware #3462: attempting to debug firmware using visual studio code generates error:  Failed to resolve Blackmagic hostname

jpitz31, 2024-02-19:
### Describe the bug.

After installing tool chain with ./fbt vscode_dist,  and setting up Flipperzero dev board acording to: https://docs.flipper.net/development/hardware/wifi-developer-board/update
Selecting Attach FW (Blackmagic) in the debug tab.  and selecting the green start debug link
generates the following error:
Failed to resolve Blackmagic hostname
scons: *** [phony_get_blackmagic] Please specify BLACKMAGIC=...

********** FBT ERRORS **********
phony_get_blackmagic: Please specify BLACKMAGIC=...

if you run ./fbt get_blackmagic you get the same error.

### Reproduction

1. Install firmware using git clone --recursive 
2. Install tool chain using ./fbt vscode_dist
3. run ./fbt, generates firmware
4. Install blackmagic on flipperzero dev board according to https://docs.flipper.net/development/hardware/wifi-developer-board/update
5. Selecting Attach FW (Blackmagic) in the debug tab.  and selecting the green start debug link
6. or running ./fbt get_blackmagic

produces the above error message:

No documentation explains how to address this issue:
Enclosed is my launch.json file

// Use IntelliSense to learn about possible attributes.
    // Hover to view descriptions of existing attributes.
    // For more information, visit: https://go.microsoft.com/fwlink/?linkid=830387
    "version": "0.2.0",
    "inputs": [
        {
            "id": "BLACKMAGIC",
            "type": "command",
            "command": "shellCommand.execute",
            "args": {
                "useSingleResult": true,
                "env": {
                    "PATH": "${workspaceFolder}${command:extension.commandvariable.envListSep}${env:PATH}"
                },
                "command": "fbt -s get_blackmagic",
                "description": "Get Blackmagic device"
            }
        }
    ],
    "configurations": [
        {
            "name": "Attach FW (ST-Link)",
            "cwd": "${workspaceFolder}",
            "executable": "./build/latest/firmware.elf",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "stlink",
            "svdFile": "./scripts/debug/STM32WB55_CM4.svd",
            // If you're debugging early in the boot process, before OS scheduler is running,
            // you have to comment out the following line.
            "rtos": "FreeRTOS",
            "configFiles": [
                "interface/stlink.cfg",
                "./scripts/debug/stm32wbx.cfg",
            ],
            "postAttachCommands": [
                "source scripts/debug/flipperversion.py",
                "fw-version",
                // "compare-sections",
                "source scripts/debug/flipperapps.py",
                "fap-set-debug-elf-root build/latest/.extapps",
                // "source scripts/debug/FreeRTOS/FreeRTOS.py",
            ]
            // "showDevDebugOutput": "raw",
        },
        {
            "name": "Attach FW (blackmagic)",
            "cwd": "${workspaceFolder}",
            "executable": "./build/latest/firmware.elf",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "external",
            "gdbTarget": "${input:BLACKMAGIC}",
            "svdFile": "./scripts/debug/STM32WB55_CM4.svd",
            "rtos": "FreeRTOS",
            "postAttachCommands": [
                "monitor swdp_scan",
                "attach 1",
                "set confirm off",
                "set mem inaccessible-by-default off",
                "source scripts/debug/flipperversion.py",
                "fw-version",
                "source scripts/debug/flipperapps.py",
                "fap-set-debug-elf-root build/latest/.extapps",
                // "compare-sections",
            ]
            // "showDevDebugOutput": "raw",
        },
        {
            "name": "Attach FW (JLink)",
            "cwd": "${workspaceFolder}",
            "executable": "./build/latest/firmware.elf",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "jlink",
            "interface": "swd",
            "device": "STM32WB55RG",
            "svdFile": "./scripts/debug/STM32WB55_CM4.svd",
            "rtos": "FreeRTOS",
            "postAttachCommands": [
                "source scripts/debug/flipperversion.py",
                "fw-version",
                "source scripts/debug/flipperapps.py",
                "fap-set-debug-elf-root build/latest/.extapps",
            ]
            // "showDevDebugOutput": "raw",
        },
        {
            "name": "Attach FW (DAP)",
            "cwd": "${workspaceFolder}",
            "executable": "./build/latest/firmware.elf",
            "request": "attach",
            "type": "cortex-debug",
            "servertype": "openocd",
            "device": "cmsis-dap",
            "svdFile": "./scripts/debug/STM32WB55_CM4.svd",
            "rtos": "FreeRTOS",
            "configFiles": [
                "interface/cmsis-dap.cfg",
                "./scripts/debug/stm32wbx.cfg",
            ],
            "postAttachCommands": [
                "source scripts/debug/flipperversion.py",
                "fw-version",
                "source scripts/debug/flipperapps.py",
                "fap-set-debug-elf-root build/latest/.extapps",
            ],
            // "showDevDebugOutput": "raw",
        },
        {
            "name": "python debug",
            "type": "python",
            "request": "launch",
            "program": "${file}",
            "args": []
        }
    ]
}

### Target

Flipperzero with 0.98.3 firmware

hedger, 2024-02-19:
That error means fbt cannot discover your blackmagic probe in LAN or over USB. How is it attached, over USB or WiFi? Which OS are you using?

jpitz31, 2024-02-19:
Hey hedger,  using flipperzero dev board connected by USB.  Dev board is
connected directly to FlipperZero. if I use pio device list I can see the
USB connections.
I am using kali linux 2023.

Thanks
Joe

-- 
http://joestechblog.com

*//
--------------------------------------------------------------------------------------*
*Old C programmers never die, they just get a new function pointer.*

*..."It began upon the following occasion: It is allowed on all hands, that
the primitive way of breaking eggs before we eat them, was upon the larger
end: but his present Majesty's grandfather, while he was a boy, going to
eat an egg, and breaking it according to the ancient practice, happened to
cut one of his fingers. Whereupon the Emperor his father published an
edict, commanding all his subjects, upon great penalties, to break the
smaller end of their eggs."  Jonathan Swift*

jpitz31, 2024-02-19:
Hey hedger,  I tried again disconnecting and reconnecting the dev board and resetting the USB port.  It seems to be working now.

Please close the ticket.

Thanks for the assistance.

Joe

hedger, 2024-02-19:
Glad to help.

closed 2024-02-19 by hedger