flipperdevices/flipperzero-firmware #3465: Parser st25tb

KaliStudio, 2024-02-19:
### Description of the feature you're suggesting.

is it possible to have an example to parse data from a st25tb card please? 
for example being able to detect if UID starts with D0 02 then retrieve the information from block 5 and display them: furi_string_cat_printf(parsed_data, "block 5: %d\n", app.block5);

gornekich, 2024-02-20:
Hello @KaliStudio 
Have a look on mykey plugin. This card works on st25tb protocol
https://github.com/flipperdevices/flipperzero-firmware/blob/dev/applications/main/nfc/plugins/supported_cards/mykey.c

KaliStudio, 2024-02-20:
> Bonjour@KaliStudio 
Okay, thanks for this information, I'll take a look!

KaliStudio, 2024-02-21:
does anyone have any mykey dumps?

closed 2024-03-05 by skotopes