flipperdevices/flipperzero-firmware #3473: Error compiling in Windows

eried, 2024-02-22:
### Describe the bug.

Hi, I am getting a strange error, is this something wrong with scons? :S

### Reproduction

1. Clone the repo
2. Run ./fbt

### Target

Windows 10

### Logs

```Text
2024-02-22 15:13:53,211 [INFO] Packing
scons: *** [build\f7-firmware-D\appframe\app-scened-template\view_modules\byte_input_vm.o] Error 1
        CC      lib\ble_profile\extra_profiles\hid_profile.c
scons: *** [build\f7-firmware-D\appframe\app-scened-template\view_modules\dialog_ex_vm.o] Error 1
r 1
scons: *** [build\f7-firmware-D\appframe\app-scened-template\view_modules\text_input_vm.o] Error 1
scons: *** [build\f7-firmware-D\appframe\app-scened-template\text_store.o] Error 1
scons: *** [build\f7-firmware-D\bit_lib\bit_lib.o] Error 1
ccache: error: The USERPROFILE environment variable must be set to your user profile folder
scons: *** [build\f7-firmware-D\ble_profile\extra_profiles\hid_profile.o] Error 1
2024-02-22 15:13:53,495 [INFO] Complete
2024-02-22 15:13:53,541 [INFO] Complete

********** FBT ERRORS **********
build\f7-firmware-D\appframe\app-scened-template\view_modules\byte_input_vm.o: Error 1
build\f7-firmware-D\appframe\app-scened-template\view_modules\dialog_ex_vm.o: Error 1
build\f7-firmware-D\appframe\app-scened-template\view_modules\popup_vm.o: Error 1
```
```

hedger, 2024-02-22:
You have a misconfigured ccache installation. Either fix it or uninstall it.

hedger, 2024-02-22:
Alternatively, please try this branch: https://github.com/flipperdevices/flipperzero-firmware/pull/3451

I implemented a potential workaround for that issue - please confirm if it helps.

eried, 2024-02-23:
> Alternatively, please try this branch: #3451

With this one is the same error:
```

scons: *** [build\f7-firmware-D\appframe\app-scened-template\text_store.o] Error 1
scons: *** [build\f7-firmware-D\bit_lib\bit_lib.o] Error 1
ccache: error: could not find configuration file and the LOCALAPPDATA environment variable is not set
scons: *** [build\f7-firmware-D\ble_profile\extra_profiles\hid_profile.o] Error 1
2024-02-23 11:12:47,048 [INFO] Complete
2024-02-23 11:12:47,074 [INFO] Complete

********** FBT ERRORS **********
build\f7-firmware-D\appframe\app-scened-template\view_modules\byte_input_vm.o: Error 1
build\f7-firmware-D\appframe\app-scened-template\view_modules\dialog_ex_vm.o: Error 1
```

But you were right about ccache :) it was Perl

[image]

closed 2024-02-23 by eried

hedger, 2024-02-23:
Aha, this time it's a different variable. I'll set up ccache on Windows to investigate that.

hedger, 2024-02-26:
@eried could you please check latest dev for that issue?

eried, 2024-02-26:
yeah it works fine now :)