flipperdevices/flipperzero-firmware #3478: Archive: Unable to rename file to name with different capitalisation

tiftia, 2024-02-24:
### Describe the bug.

Archive does not allow renaming file to the same name with a different capitalisation, probably because it includes the renamed file name in the list of already used file names.

### Reproduction

1. Open archive
3. Rename some file changing only the case of some letters
4. Click save
5. Archive prints: "This name exists! Choose another name.".

hedger, 2024-02-26:
For FAT family filesystems, names with different capitalisation are considered equal. 
You can rename the file to a temporary name, then back to the desired one.

CookiePLMonster, 2024-03-01:
Branched this issue off to a new ticket: https://github.com/flipperdevices/flipperzero-firmware/issues/3492.