flipperdevices/flipperzero-firmware #3481: Save All Received Messages

imply3289, 2024-02-25:
### Description of the feature you're suggesting.

When using the 'Read' function of the inbuilt SubGHz application, it would be helpful to be able to with one action, save _all_ the recorded messages.

The name for each message could then be the same name, just with an incrementing suffix.
For example, if the user entered 'remote' as the name, the resultant saves would be named 'remote01', 'remote02', and so on up to the 50 that it can capture at a time.

### Anything else?

This would be especially helpful for learning about systems that are repetitive but share a common underlying protocol structure.

For example, a restaurant pager system. If you had access to a system to learn about, you could trigger all the individual pagers in order, then save all those captures in one go, to be later studied individually to learn how each message is unique or how the pager numbers are encoded into the protocol.

mcauser, 2024-03-03:
Alternatively, add a config option to always save recorded messages.
A "recon mode" where they are saved to SD as soon as they are heard, with the predefined filename pattern.
Incrementing suffix would be nice, or a timestamp

skotopes, 2024-04-14:
There are no such plans to do it right now.

In general subghz got its API fully exposed to developers, so it shouldn't be a problem to make such app.

closed 2024-04-14 by skotopes