flipperdevices/flipperzero-firmware #3483: Flipper crashed after getting info of NFC-V tag

mxcdoam, 2024-02-26:
### Describe the bug.

Flipper Zero randomly crashes with error "NULL pointer dereference" after reading several "ISO 15693-3 (Unknown)" tags (NXP TagInfo recognizes them as ICODE SLIX  SL2S2002) and attempting to get tag's info.  This bug appears randomly, to capture it on video i had 4 reading/getting info attempts. 
FW is 0.98.3
Video is linked below.
[https://www.youtube.com/shorts/CS9_5MMvPFo](url)

### Reproduction

1) Read several ICODE SLIX tags without exiting NFC app
2) Get info of read tags.
2) After 3rd-7th tag is read , open menu and select "Info". 
Repeat until the crash occurs.

skotopes, 2024-03-05:
@mxcdoam can you also reproduce this issues with latest rc or dev?

mxcdoam, 2024-03-05:
> @mxcdoam can you also reproduce this issues with latest rc or dev?

I can't reproduce this issue on 0.99.0-rc, so it seems to be fixed.

skotopes, 2024-03-05:
awesome ) then I'm closing issue

closed 2024-03-05 by skotopes