flipperdevices/flipperzero-firmware #3487: Wifi board not found

bitsandscience, 2024-02-28:
### Describe the bug.

I got recently a Wifi devboard and following all steps in the documentation in FZ website it generates the following error in Windows 10: 
scons: Entering directory `C:\Users\Home\.ufbt\current\scripts\ufbt'
←[1;35m
fbt: warning: App folder 'C:\Users\Home': missing manifest (application.fam)
←[0m←[33m←[33mLoadAppManifest, line 130, ←[0min file "C:\Users\Home\.ufbt\current\scripts\fbt_tools\fbt_apps.py"
←[0mpython C:\Users\Home\.ufbt\current\scripts/wifi_board.py
2024-02-28 14:19:15,857 [ERROR] WiFi board not found
2024-02-28 14:19:15,858 [INFO] Please connect WiFi board to your computer, hold down BOOT button and press RESET button
scons: *** [phony_devboard_flash] Error 1

I have tried installing ufbt 0.2.5 and ufbt 0.2.4.3 after uninstalling the latest one. I also see that drivers are not installed in device manager but following espressif instructions doesn't help a lot (here: https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32s2/get-started/establish-serial-connection.html) Steps to reproduce from 4 on are for trying to install drivers. I set up the COM with 115200-8-1-N and flow control =Xon/Xoff.

### Reproduction

1. pip install ufbt
2. Plugging wifi board to PC while holding boot button, then pressing reset and releasing both (also holding boot button after plugging).
3. python -m ufbt devboard_flash (as stated but without '3' in docs).
4. Set up COM as said in espressif doc, open session in putty with same parameters.
5. Press reset button (fatal error occurs).

hedger, 2024-02-28:
We've just made an experimental fix for certain Windows-related devboard detection issues. It's #3485.

Please use latest ufbt version and install `zlo/wifi_board_update_windows_fix` branch of the firmware's SDK:
`python -m ufbt update -b zlo/wifi_board_update_windows_fix` or `ufbt update -b zlo/wifi_board_update_windows_fix`

Then retry the devboard update with `ufbt devboard_flash`.

bitsandscience, 2024-02-28:
Many thanks! It worked!

Health-and-Public-Sector, 2024-03-03:
C:\Windows\System32>ufbt update -b zlo/wifi_board_update_windows_fix
19:25:58.612 [I] Deploying SDK for f7
19:25:58.612 [I] Fetching branch index https://update.flipperzero.one/builds/firmware/zlo/wifi_board_update_windows_fix/
19:25:58.827 [E] Failed to run operation: HTTP Error 404: Not Found. See --verbose for details

--------------------------------------------------------------------------------------------------------------------

C:\Windows\System32>python -m ufbt update -b zlo/wifi_board_update_windows_fix
19:26:27.279 [I] Deploying SDK for f7
19:26:27.279 [I] Fetching branch index https://update.flipperzero.one/builds/firmware/zlo/wifi_board_update_windows_fix/
19:26:27.405 [E] Failed to run operation: HTTP Error 404: Not Found. See --verbose for details

PR #3485 (2024-03-01): WiFi board: fixed update script on Windows (unfortunately also Mac and Linux)
# What's new

- WiFi board: fixed update script
- fbt: dev channel for devboard flash
- fbt/ufbt: added ARGS parameter for certain scripts
- fixes #3487, #3222

# Verification 

- `./fbt devboard_flash`
- `./fbt devboard_flash ARGS="-c dev"`

# Checklist (For Reviewer)

- [ ] PR has description of feature/bug or link to Confluence/Jira task
- [ ] Description contains actions to verify feature/bugfix
- [ ] I've built this code, uploaded it to the device and verified feature/bugfix