flipperdevices/flipperzero-firmware #3493: Windows toolchain is missing python pillow package

Kingwad, 2024-03-02:
### Describe the bug.

The Windows toolchain is missing the python Pillow package required to build the updater_package target. The toolchain is supposed to include all python packages to alleviate the need to install packages system-wide.

### Reproduction

On a Windows machine that does not have Pillow installed in the system level python pip packages:

Run `./fbt COMPACT=0 DEBUG=1 VERBOSE=0 updater_package copro_dist`

See that it fails to build with

````
Traceback (most recent call last):
  File "C:\Users\adam\dev\flipperzero-firmware\applications_user\flipperzero-game-engine-example\engine\scripts\sprite_builder.py", line 8, in <module>
    from PIL import Image, ImageOps
ModuleNotFoundError: No module named 'PIL'
```

### Logs

```Text
PS C:\Users\adam\dev\flipperzero-firmware> ./fbt COMPACT=0 DEBUG=1 VERBOSE=0 updater_package
        SDKCHK  targets\f7\api_symbols.csv
API version 58.0 is up to date
Linking build\f7-firmware-D as latest built dir (./build/latest/)
        HEX     build\f7-updater-D\updater.hex
        BIN     build\f7-updater-D\updater.bin
updater.bin:   29 flash pages (last page 64.77% full)
        EXTCMD  applications_user\flipperzero-game-engine-example\assets
        APPMETA build\f7-firmware-D\.extapps\microel_parser.fal
        FAP     build\f7-firmware-D\.extapps\microel_parser.fal
        APPMETA build\f7-firmware-D\.extapps\mizip_parser.fal
        FASTFAP build\f7-firmware-D\.extapps\microel_parser.fal
        FAP     build\f7-firmware-D\.extapps\mizip_parser.fal
        DFU     build\f7-updater-D\updater.dfu
        CDB     build\f7-firmware-D\compile_commands.json
        FASTFAP build\f7-firmware-D\.extapps\mizip_parser.fal
        APPMETA build\f7-firmware-D\.extapps\hi_parser.fal
        FAP     build\f7-firmware-D\.extapps\hi_parser.fal
Traceback (most recent call last):
  File "C:\Users\adam\dev\flipperzero-firmware\applications_user\flipperzero-game-engine-example\engine\scripts\sprite_builder.py", line 8, in <module>
    from PIL import Image, ImageOps
ModuleNotFoundError: No module named 'PIL'
        FASTFAP build\f7-firmware-D\.extapps\hi_parser.fal
        APPMETA build\f7-firmware-D\.extapps\opal_parser.fal
        FAP     build\f7-firmware-D\.extapps\opal_parser.fal
        FASTFAP build\f7-firmware-D\.extapps\opal_parser.fal
        APPMETA build\f7-firmware-D\.extapps\mykey_parser.fal
        FAP     build\f7-firmware-D\.extapps\mykey_parser.fal
        FASTFAP build\f7-firmware-D\.extapps\mykey_parser.fal
        APPMETA build\f7-firmware-D\.extapps\myki_parser.fal
        FAP     build\f7-firmware-D\.extapps\myki_parser.fal
        APPMETA build\f7-firmware-D\.extapps\troika_parser.fal
        FASTFAP build\f7-firmware-D\.extapps\myki_parser.fal
        FAP     build\f7-firmware-D\.extapps\troika_parser.fal
        APPMETA build\f7-firmware-D\.extapps\washcity_parser.fal
        FAP     build\f7-firmware-D\.extapps\washcity_parser.fal
        FASTFAP build\f7-firmware-D\.extapps\troika_parser.fal
        LINK    build\f7-firmware-D\.extapps\plantain_parser_d.elf
        FASTFAP build\f7-firmware-D\.extapps\washcity_parser.fal
        LINK    build\f7-firmware-D\.extapps\two_cities_parser_d.elf
        LINK    build\f7-firmware-D\.extapps\aime_parser_d.elf
        LINK    build\f7-firmware-D\.extapps\umarsh_parser_d.elf
scons: *** [applications_user\flipperzero-game-engine-example\assets] Error 1
        CC      applications\main\nfc\plugins\supported_cards\gallagher.c

********** FBT ERRORS **********
applications_user\flipperzero-game-engine-example\assets: Error 1
```

Kingwad, 2024-03-02:
Tested on release 98.3 (toolchain 23) and dev (toolchain 33) and the result is the same

skotopes, 2024-03-05:
@hedger @DrZlo13 FYI

Willy-JL, 2024-03-06:
the issue is that this app uses compile command with `python3` which is not part of toolchain on windows, so it uses the system python, which does not have pil.

solution is as per https://github.com/flipperdevices/flipperzero-good-faps/pull/155

skotopes, 2024-03-07:
We'll update flipperzero-game-engine-example

DrZlo13, 2024-03-07:
Air Arkanoid and flipperzero-game-engine-example have been updated to use python from the toolchain.

closed 2024-03-07 by DrZlo13