flipperdevices/flipperzero-firmware #3494: Keri RFID NXT not woring

devopgirlie, 2024-03-04:
Keri key fob NXT series (starts with N before serial number) does not read under RFID or NFC. Unable to get a RAW read as well. Is there a fix to this issue? Thanks in advance

skotopes, 2024-03-05:
@devopgirlie where can we find those fobs? are they on sale anywhere?

@doomwastaken can we get some?

doomwastaken, 2024-03-05:
Yeah, I found them for another ticket a while back, will order them

devopgirlie, 2024-03-06:
@skotopes @doomwastaken found them here https://www.idzone.com/nxt-k-pf.html

here are the specs from the site: https://kerisys.com/products/nxt-credentials/

doomwastaken, 2024-03-06:
The one I had in mind is https://www.dkhardware.com/nxt-k-keri-systems-proxfob-nxt-k-product-1138628.html

amanbaissa, 2024-08-26:
@doomwastaken has there been an update on reading NXT keri cards?

Delso411, 2024-08-30:
It has been no update to my knowledge i always try once in a while to copy keri nxt series key fob which is my main keyfob and i had lost one and been trying to copy one for the longest

doomwastaken, 2024-08-30:
We had recently ordered key fobs, sadly I was unable to find them in the UK. Once they arrive, we will get back to you. Sorry for delay

skotopes, 2024-09-05:
@devopgirlie just arrived, we'll take a look on this week

0xDegenDeveloper, 2024-09-12:
Looking forward to an update on this, been trying to copy my sauna's fob for 2+ years now. Its a Keri NX... tag, with all the firmwares I've had no luck. 

Curious if we expect the NX tags to be possible with a flipper, or if I should look into other devices to do so.

skotopes, 2024-09-13:
Those tags are indeed quite unusual. We'll need more time to add their support.

OnlineMartialArtist, 2024-11-07:
I couldn't get the flipper to read my Keri RFID card, but Keysey was able to read it and clone it. Keysey is a small device for copying/cloning/erasing an RFID card and it's sold on Amazon. To this day, I can't get the flipper to read my Keri fob. 🤷‍♂️

forxhunter, 2024-12-09:
any progress?

OnlineMartialArtist, 2025-01-01:
No progress yet, but I’ve been working on it a lot recently because someone in my building just got a flipper and we’ve been tinkering/researching constantly. I know someone made this script for raw files to RFID in python, but I haven’t tried it yet. I tried the plotter for the raw files and it didn’t give me anything I could see that was usable to create a key manually for Keri in hex (which is now an option). 

here’s the link for the python code:

https://github.com/hnesk/flipper-raw-rfid

PLEASE ADD SUPPORT FOR KERI NXT RFID! Ty!

RetailCircle, 2025-01-04:
I couldn't read and clone my KERI NXT-K fob today either. If someone has a solution, please let me know.

JJazz-jpg, 2025-03-21:
Interested in support being added for this as well

amanbaissa, 2025-05-06:
@skotopes any updates to reading the NXT fobs?