flipperdevices/flipperzero-firmware #3498: weekday not set in datetime_timestamp_to_datetime

mdaskalov, 2024-03-06:
### Describe the bug.

In the function `datetime_timestamp_to_datetime` (in the file `/lib/datatime/datatime.c` of release `0.99.0-rc`) the `datatime->weekday` is left unchanged / uninitialized.

It is also present in the function `furi_hal_rtc_datetime_to_timestamp` of the current release (`0.98.3`) but in different file: `fury_hal_rtc.c`.

You can use following to update it (01.01.1970 was a Thursday):

```
datatime->weekday = ((days + 4) % 7) + 1;
``` 

### Reproduction

Exemple test code:
``` 
    uint32_t test_value = 1709748421;

    DateTime datetime = {0};
    datetime_timestamp_to_datetime(test_value, &datetime);

    mu_assert_int_eq(datetime.weekday, 3);
```

### Target

f7-firmware-D

CookiePLMonster, 2024-03-06:
To add to that, the current layout of `DateTime` is sub-optimal, with the size of 10 bytes. Had `weekday` been moved ahead of `year`, it'd fit in the padding added there to align `year` to 2 bytes.

Consider moving `weekday` before `day`, so it stays grouped logically while also not generating 2 bytes of padding.

skotopes, 2024-03-11:
struct packing is coming later

PR #3508 (2024-03-11): DateTime: fix missing weekday in datetime_timestamp_to_datetime
# What's new

- DateTime: fix missing weekday in datetime_timestamp_to_datetime
- Fixes #3498

# Verification 

- Check unit tests

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix