flipperdevices/flipperzero-firmware #3501: How to correctly include crypto1.h?

zWaR, 2024-03-07:
Hey there,

I'm compiling a custom application for Flipper Zero. I'm using `ufbt`. The code is including `crypto1.h` like his:

```c
#include <lib/nfc/protocols/mf_classic/crypto1.h>
```

I made sure that this include follows the path to `crypto1.h` in the `dev` branch of the official firmware repository on GitHub: https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/nfc/protocols/mf_classic/crypto1.h

I made sure `ufbt` is using firmware's latest `dev` branch by running:

```shell
ufbt update --channel=dev
```

However, when running `ufbt`, I get the following error:

```shell
fatal error: lib/nfc/protocols/mf_classic/crypto1.h: No such file or directory
3 | #include <lib/nfc/protocols/mf_classic/crypto1.h>
```

What's the correct way to include `crypto1.h`?

Thanks.

CookiePLMonster, 2024-03-07:
This header is not exposed as public:
https://raw.githubusercontent.com/flipperdevices/flipperzero-firmware/dev/targets/f7/api_symbols.csv

Multiple `lib/nfc/protocols/mf_classic` headers are exposed, but crypto isn't one of them.

zWaR, 2024-03-07:
Oh, makes sense. Thanks @CookiePLMonster!

closed 2024-03-07 by zWaR