flipperdevices/flipperzero-firmware #3509: Video Game Module not initialized

otanim, 2024-03-11:
### Describe the bug.

Video Game Module is not being recognized by Flipper Zero when it's plugged in. Nothing happens, no notifications are popping up on Flipper Zero. When HDMI is plugged in to Video Game Module, it shows a small screen "Video Game Module is not initialized".

The Firmware version: 0.98.3 (official)

### Reproduction

1. Connect Video Game Module to Flipper Zero.

skotopes, 2024-03-11:
- Check device Settings: `Expansion Modules` -> `Listen UART` must be set to `USART`
- Disconnect VGM, reboot flipper with `Left` + `Back` key combo and reconnect VGM

If you still experiencing issue: contact support at https://support.flipper.net/hc/en-us

otanim, 2024-03-11:
Thanks @skotopes, I'm all set now.
Sorry, should've read the docs attentively, your instruction was already been included in [docs](https://docs.flipper.net/video-game-module/quick-start#dX4LQ):
[image]