flipperdevices/flipperzero-firmware #3525: Using iso14443_3a_poller_txrx_custom_parity with MfClassicPoller

bettse, 2024-03-20:
I've got a `MfClassicPoller`, but I want to use `iso14443_3a_poller_txrx_custom_parity` (as seen in `protocols/mf_classic/mf_classic_poller_i.c`), but I need to pass a `Iso14443_3aPoller` and the `MfClassicPoller` is an incomplete definition (opaque) so I can't just do `mfc_poller->iso14443_3a_poller`.
Am I doing it totally wrong, or is there a way, or do we need a `get_iso14443_3a_poller(MfClassicPoller *mfc_poller)`?

gornekich, 2024-03-28:
Hello @bettse . Could you please tell more about what you want to do?

I can see the problem that if we want to send custom MfClassic messages on Iso14443-3a events, we don't have this opportunity yet. I think the best solution here is to add mf_classic_poller_send_frame() function, which must be called in callback, and will transmit and receive frames in accordance with Crypto1 state inside MfClassicPoller.

bettse, 2024-03-28:
I have a device that constructs the encrypted mifare classic message it would like to have sent.  I used `FuriHalNfcTxRxTypeRaw` in the pre-refactor situation (https://github.com/bettse/seader/pull/2/files#diff-c5d1f77597b28bba866e063cd68bcbd246650e4b82d7a332ce99af86978b5894R572)

(That PR might be enough context, but let me now if you'd like more)

bettse, 2024-04-13:
Anything on this?  Is more info needed?

gornekich, 2024-04-15:
@bettse sorry for late response. Added some function in #3576 . Once it's merged, you can start poller with MfClassic protocol and send all types of frames in callback. You can take mf_classic_send_frame_test() function in unit test as an example. I believe that would be enough to resolve your issue.

gornekich, 2024-04-16:
Please, reopen the issue if you still have some problems with MFC API

closed 2024-04-16 by gornekich