flipperdevices/flipperzero-firmware #3531: UART transfer data error

EstebanFuentealba, 2024-03-21:
### Describe the bug.

I recently updated my FAPs to the latest version, and they worked perfectly. I heavily utilize UART and data transfer, and I also updated to the latest DateTime. But 🤔 I've encountered issues with the build generated with the latest version. I installed the FAP from the catalog (weighing 55kb), but when transferring via UART in my app, it gives a "[Parse Error](https://github.com/EstebanFuentealba/MALVEKE-Flipper-Zero/blob/fb7dba8c813a659c81fc37e5e7150c0b939fd4a7/flipper_companion_apps/applications/external/malveke_gb_cartridge/views/gb_cartridge_scene_1.c#L43)" (When I transfer a JSON from ESP32 to FZ, it means that it couldn't be deserialized). Perhaps this is related to some new changes in furi_stream_buffer, because in a FAP generated weeks earlier (60kb), it works perfectly fine.

**Fap From Catalog**
[malveke_gb_cartridge.fap.zip](https://github.com/flipperdevices/flipper-application-catalog/files/14695227/malveke_gb_cartridge.fap.zip)
[image]

**Fap generated weeks earlier**

[malveke_gb_cartridge.fap.zip](https://github.com/flipperdevices/flipper-application-catalog/files/14695262/malveke_gb_cartridge.fap.zip)

[image]

Both FAPs were run on the same firmware (latest version).

[image]

https://github.com/EstebanFuentealba/MALVEKE-Flipper-Zero/issues/31

Command And Response
[image]

### Reproduction

1. Install fap from catalog
2. Open app
3. Select "Cartridge Information"
4. Obtain Error

### Target

0.99.1

### Logs

```Text
This is what the ESP32 sends once it reads the GAME BOY cartridge: a string that I deserialize as JSON using cJSON.

`JSON:{"type":"info","gb_type":"gbc","gb_sgb":true,"logo":[206,237,102,102,204,13,0,11,3,115,0,131,0,12,0,13,0,8,17,31,136,137,0,14,220,204,110,230,221,221,217,153,187,187,103,99,110,14,236,204,221,220,153,159,187,185,51,62],"logo_correct":1,"success":1,"message":"","checksum":"047C","title":"POKEMON YELLOW","isCamera":false,"ramBanks":4,"ramEndAddress":49151,"romBanks":64,"romEndAddress":32767,"rev":0,"mapper":"MBC5","ROMSize":"1 MB","RAMSize":"32 KB"}`
```

skotopes, 2024-03-21:
Sadly we don't have this module to reproduce this issue.

Let's try to start from the other side:
- Which API is misbehaving?
- Can you show us your code that misbehaves?
- Can you make simple code example that we can test without additional hardware?

EstebanFuentealba, 2024-03-21:
I will create a demo.ino file that can be flashed onto any ESP32 board with the bug.

EstebanFuentealba, 2024-03-21:
This is a simple .ino code that can be flashed onto any ESP32.

[image]

```ino
void setup() {
  // put your setup code here, to run once:
  Serial.begin(115200);
  Serial.println("Bug: #3531");
}
String getSerialInput() {
  String input = "";

  if (Serial.available() > 0)
    input = Serial.readStringUntil('\n');

  input.trim();
  return input;
}

void loop() {
  String input = getSerialInput();
  if (input == "gbcartridge -i") {
    Serial.println("JSON:{\"type\":\"info\",\"gb_type\":\"gbc\",\"gb_sgb\":true,\"logo\":[206,237,102,102,204,13,0,11,3,115,0,131,0,12,0,13,0,8,17,31,136,137,0,14,220,204,110,230,221,221,217,153,187,187,103,99,110,14,236,204,221,220,153,159,187,185,51,62],\"logo_correct\":1,\"success\":1,\"message\":\"\",\"checksum\":\"047C\",\"title\":\"POKEMON YELLOW\",\"isCamera\":false,\"ramBanks\":4,\"ramEndAddress\":49151,\"romBanks\":64,\"romEndAddress\":32767,\"rev\":0,\"mapper\":\"MBC5\",\"ROMSize\":\"1 MB\",\"RAMSize\":\"32 KB\"}");
  }
}
```

**Video Demo**: 
[[image]](https://youtu.be/yl29lCPHQ7E)

skotopes, 2024-03-29:
@EstebanFuentealba I've took a look at your code: you are using `FuriHalSerialRxEventIdle` incorrectly. You must not read data if you are using async_rx and receiving idle event because there is nothing to read. Use furi_hal_serial_async_rx  only with FuriHalSerialRxEventData event.

However if you are using `furi_hal_serial_dma_rx_start` then Idle also must be used for reading.

skotopes, 2024-03-29:
Please let me know if it fixes your issue.

EstebanFuentealba, 2024-03-29:
I made the change by removing `FuriHalSerialRxEventIdle`, but it didn't work.

[image]

I also switched to using the method `furi_hal_serial_dma_rx_start`, but that didn't work either. 
[image]

I'm going to clone the firmware and search for the commit where it was last working.

EstebanFuentealba, 2024-03-30:
🤔 I cloned the Flipper Zero repository and placed the code for my app in the directory 
`applications_user/malveke_gb_cartridge`. 

I began testing commit by commit on the **dev** branch using 
`./fbt launch APPSRC=malveke_gb_cartridge`
The app worked correctly in all of them. I also tested the latest commit on the **release** branch, and it worked as well.

I then conducted a test with the latest commit from the release branch but this time with UFBT. I navigated to the directory `cd applications_user/malveke_gb_cartridge && ufbt launch`
which resulted in a "[Parse Error](https://github.com/EstebanFuentealba/MALVEKE-Flipper-Zero/blob/fb7dba8c813a659c81fc37e5e7150c0b939fd4a7/flipper_companion_apps/applications/external/malveke_gb_cartridge/views/gb_cartridge_scene_1.c#L43)" again. Could it be an issue with UFBT?

Application generated with **FBT**.
[image]

Application generated with **UFBT**.
[image]

EstebanFuentealba, 2024-04-03:
Hi @skotopes  I have been comparing the `malveke_gb_cartridge.impsyms` files generated by **ufbt** and **fbt**, and in the one generated by **ufbt**, `strcpy U` is added.

```diff
--- /Users/estebanfuentealba/flipperzero-firmware/build/f7-firmware-D/.extapps/malveke_gb_cartridge.impsyms	2024-04-03 09:03:12
+++ /Users/estebanfuentealba/.ufbt/build/malveke_gb_cartridge.impsyms	2024-04-03 09:04:44
@@ -95,6 +95,7 @@
 storage_file_write U         
 storage_simply_mkdir U         
 strcmp U         
+strcpy U         
 strdup U         
 strlen U         
 strncmp U         
```

With this information, I checked the compilation flags used by **ufbt**, and in the `sdk_headers/sdk.opts` for release, it has `-Os`, which is compiler optimization, perhaps removing some functions (like `strcpy`) from `string.h` that my code uses.

I also tried compiling with `-O1` (Basic optimization level), and the FAP works 🤓.

I've been searching, and perhaps changing `-Os` to `-O1` is most likely to work.

https://github.com/flipperdevices/flipperzero-firmware/blob/d32f1955578782efd2c9735476d8c6048c54043f/firmware.scons#L44

https://github.com/flipperdevices/flipperzero-firmware/blob/d32f1955578782efd2c9735476d8c6048c54043f/firmware.scons#L56

I'm not sure what the specific reason might be for the optimization to affect some functions 🤔.

closed 2024-04-04 by doomwastaken