flipperdevices/flipperzero-firmware #3534: API request: Free heap blocks available (same as memmgr_heap_printf_free_blocks)

noproto, 2024-03-21:
### Description of the feature you're suggesting.

Currently there's no way to check all of the available heap blocks of memory via the API.

It's not possible to plan heap allocation without repeatedly reserving memory and checking the next largest block with memmgr_heap_get_max_free_block (and if there's insufficient memory rolling back all of the allocations and re-allocating).

The only way I can see (which is a hack) is by setting up a log handler with furi_log_add_handler, call memmgr_heap_printf_free_blocks to dump the free blocks to the log buffered output stream, read back the log and parse each line, then use that in any heap memory planning function.

It would be very helpful to expose a function which can be used to plan help allocation/memory management in FAPs. If the largest sized block isn't checked, any malloc can fail freezing the device.

CookiePLMonster, 2024-03-21:
Would this idea accomplish much? There are many services running on the device at any point and FreeRTOS employs dynamic allocation, so if you used the result of such a function to plan your allocations, you could run into a race where the block(s) were free when you checked, but they stopped being free when you are allocating.

Sounds like it'd be better to guarantee that a failed malloc is "safe" and returns null instead of freezing/crashing.

noproto, 2024-03-21:
Both ideas aren't mutually exclusive. This is a low LoE change that will, regardless of possible race conditions, significantly reduce the number of crashes that users experience running my application. In my tests I haven't observed any background services consuming significant RAM unexpectedly.

DrZlo13, 2024-04-08:
@noproto I'm currently integrating new tlsf memory allocator #3572, which is much more mature and ram-friendly than our heap4 allocator. During the integration process, I created a block walker method to eliminate potential heap corruption (enumerating the heap without locking the kernel is bad, mmmkay?), and it solves exactly your problem, allows you to enumerate memory blocks.

skotopes, 2024-07-07:
Looks like TLSF is not coming )