flipperdevices/flipperzero-firmware #3537: Automatically detecting the voltage at which a module operates.

EstebanFuentealba, 2024-03-24:
### Description of the feature you're suggesting.

Is it possible to electronically implement or using the "Expansion Module Protocol," for F0 to activate 5V automatically? This would involve sending a signal to utilize `furi_hal_power_enable_otg`.

With this capability, each module could communicate to F0 the voltage at which it operates, thereby avoiding some voltage drops that occur at 3.3V, it would eliminate the need to route through `GPIO` / `5V on GPIO  ON`

skotopes, 2024-03-26:
We can not keep 5v by default on. The only viable option is to start module with 3.3V, then use RPC to enable 5V.

If that sounds good then we can expose 5v through RPC.

EstebanFuentealba, 2024-03-26:
If it sounds good, it's what I had imagined.

gsurkov, 2024-08-09:
@EstebanFuentealba 
The requested functionality is implemented in #3830.

A simpler module can use the plain expansion module protocol to enable/disable 5V on GPIO.
If a module is already using RPC over expansion module protocol, it can both enable/disable and query 5V status using the new RPC messages.

However, implementing the full RPC communication scheme only to enable 5V is not advisable and might not be feasible at all, depending on the uC used in the module.

Please refer to the [expansion module documentation](https://github.com/flipperdevices/flipperzero-firmware/blob/gsurkov/3889_5v_gpio/documentation/ExpansionModules.md) for more info.
There are also usage examples in the [Expansion module test application](https://github.com/flipperdevices/flipperzero-firmware/blob/gsurkov/3889_5v_gpio/applications/debug/expansion_test/expansion_test.c) and the [Video Game Module firmware](https://github.com/flipperdevices/video-game-module/blob/dev/app/uart.c).

closed 2024-09-05 by skotopes