flipperdevices/flipperzero-firmware #3546: Plantain parser not working properly

mxcdoam, 2024-03-27:
### Describe the bug.

Plantain parser does not work properly when trying to read MIFARE Plus X card. First, "don't move" screen appears and stays for about 20 seconds (if you try to remove card or press back button at this stage there is 50% chance that flipper will hang and will need left+back combo), then - nfc app starts dictionary attack with user and system dicts. This will take significant amount of time, so attached log is omitted.
  If you skip dict attack at a very beginning of system dict, the app will show card's type as Troika+Plantain (even if it's not) and a correct balance.
FW is 0.99.1 (but the same behaviour appears to be on 0.100.0-rc too)

### Reproduction

1. Open NFC app and put MIFARE Plus X plaintain on Flipper's back
2. Press "Read". 
3. Skip dictionary attack or watch the full cycle read with a dictionary attack, even though plantain keys are known. 

### Logs

```Text
Firmware version: 0.99.1 0.99.1 (90ae1f33 built on 14-03-2024)

>: log debug
Current log level: debug
Use <log ?> to list available log levels
Press CTRL+C to stop...
2815822 [D][NfcScanner] Found 5 base protocols
2815830 [D][DolphinState] icounter 2097, butthurt 0
2815868 [D][Nfc] FWT Timeout
2815893 [D][Nfc] FWT Timeout
2815945 [D][Nfc] FWT Timeout
2815973 [D][Nfc] FWT Timeout
2815988 [D][NfcScanner] Found 4 children
2816031 [D][Nfc] FWT Timeout
2816103 [D][Iso14443_4aPoller] Read ATS success
2816124 [I][NfcScanner] Detected 1 protocols
2816290 [I][Elf] Total size of loaded sections: 888
2816292 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816333 [I][Elf] Total size of loaded sections: 420
2816336 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816383 [I][Elf] Total size of loaded sections: 1800
2816386 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816433 [I][Elf] Total size of loaded sections: 2944
2816436 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816478 [I][Elf] Total size of loaded sections: 284
2816481 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816529 [I][Elf] Total size of loaded sections: 1356
2816532 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816574 [I][Elf] Total size of loaded sections: 820
2816577 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816619 [I][Elf] Total size of loaded sections: 552
2816622 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816680 [I][Elf] Total size of loaded sections: 1396
2816683 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816734 [I][Elf] Total size of loaded sections: 1608
2816737 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816781 [I][Elf] Total size of loaded sections: 908
2816784 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816836 [I][Elf] Total size of loaded sections: 1208
2816839 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816883 [I][Elf] Total size of loaded sections: 1320
2816886 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816933 [I][Elf] Total size of loaded sections: 1708
2816936 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2816983 [I][Elf] Total size of loaded sections: 1888
2816986 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2817032 [I][Elf] Total size of loaded sections: 1464
2817035 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2817079 [I][Elf] Total size of loaded sections: 636
2817082 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2817128 [I][Elf] Total size of loaded sections: 1024
2817131 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2817135 [D][NfcSupportedCards] Loaded 18 plugins
2817152 [D][MfClassicPoller] 4K detected
2817271 [I][MfClassicApp] Key cache not found
[... 821 lines cut ...]
2828438 [D][Microel] Entering Microel KDF
2828479 [D][Microel] UID identified: 0446932A
2828529 [I][Elf] Total size of loaded sections: 1608
2828532 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2828536 [D][MiZIP] Verifying sector 0
2828559 [D][Nfc] FWT Timeout
2828561 [D][Nfc] FWT Timeout
2828585 [D][MiZIP] Failed to read block 0: 5, this is not a MiZIP card
2828589 [D][MiZIP] Verifying sector 0
2828612 [D][Nfc] FWT Timeout
2828614 [D][Nfc] FWT Timeout
2828638 [D][MiZIP] Failed to read block 0: 5, this is not a MiZIP card
2828684 [I][Elf] Total size of loaded sections: 1708
2828687 [D][Fap] Library for NfcSupportedCardPlugin, API v. 1 loaded
2828691 [D][Plantain] Verifying sector 8
2828787 [D][MfClassicPoller] 4K detected
2828901 [D][MfClassicPoller] Auth to block 0 with key A: ffffffffffff
2828907 [D][MfClassicPoller] Reading block 0
2828912 [D][MfClassicPoller] Reading block 1
2828917 [D][MfClassicPoller] Reading block 2
2828922 [D][MfClassicPoller] Reading block 3
2828932 [D][Nfc] FWT Timeout
2828939 [D][MfClassicPoller] Auth to block 0 with key B: ffffffffffff
2828950 [D][Nfc] FWT Timeout
```

gornekich, 2024-04-15:
Hello @mxcdoam . It looks like plantain plugin can't read all the keys.

Actually, when "don't move" screen appears, the plantain plugin reads card with preset keys. However I can't see it in the logs. Could you update firmware, NFC application and provide the full logs of reading, before the dictionary attack starts?

mxcdoam, 2024-04-16:
I'm on 0.100.3.  Skipped dictionary attack, and here is the log
[MIFARE PLUS X.log](https://github.com/flipperdevices/flipperzero-firmware/files/14987468/MIFARE.PLUS.X.log)

gornekich, 2024-04-16:
@mxcdoam thanks for update. There are 2 issues here:
1) BIP plugin passes verification and tries to read card. This results in long "don't move" screen. #3595 fixes that.
2) According to your logs, Plantain parser can't read 100 block with key 3a8a139c20b4. Is it possible that you remove card from flipper by that time?

Please, update NFC app from #3595 , read card again and attach logs

mxcdoam, 2024-04-16:
> @mxcdoam thanks for update. There are 2 issues here:
>
> 1) BIP plugin passes verification and tries to read card. This results in long "don't move" screen. #3595 fixes that.
>
> 2) According to your [...]

No, absolutely not. I've tested multiple times result is always the same 

> Please, update NFC app from #3595 , read card again and attach logs here

Ok, will do that tomorrow.

gornekich, 2024-04-16:
> No, absolutely not. I've tested multiple times result is always the same. I'll also check if block 100 is protected with mentioned key.

Can you read all keys and sectors with dict attack?

mxcdoam, 2024-04-16:
> > No, absolutely not. I've tested multiple times result is always the same. I'll also check if block 100 is protected with mentioned key.
> 
> 
> 
> Can you read all keys and sectors with dict attack?

After full dictionary attack flipper shows that 69/80 keys are found and 40/40 sectors read.

mxcdoam, 2024-04-17:
> Please, update NFC app from #3595 , read card again and attach logs here

Here is a log made with dict attack skipping:
[MPX skipped dict.log](https://github.com/flipperdevices/flipperzero-firmware/files/15007088/MPX.skipped.dict.log)
Here is the log with dict attack:
[MPX with dict.log](https://github.com/flipperdevices/flipperzero-firmware/files/15007302/MPX.with.dict.log)

> 2. According to your logs, Plantain parser can't read 100 block with key 3a8a139c20b4. Is it possible that you remove card from flipper by that time?

I've checked different cards with MCT on the phone and my own dict, so it appears this key you've mentioned is protecting block 100 only on latest EV1 cards and newer PLUS X cards. Older PLUS X  (which are still in use) - all have a different key in that sector.

gornekich, 2024-04-17:
@mxcdoam thanks for testing. Can you provide the dump of older PLUS X card with all keys found? You can change the data, keys are necessary. I think we can add support to these cards to Plantain plugin

gornekich, 2024-04-17:
I prefer .nfc :)

mxcdoam, 2024-04-17:
I'll check if there is the same subset of keys on all older cards and will dump one if it is so👍

mxcdoam, 2024-04-17:
Here you go:
[Old_x_4k.zip](https://github.com/flipperdevices/flipperzero-firmware/files/15010029/Old_x_4k.zip)

gornekich, 2024-04-17:
@mxcdoam thanks a lot! Can you confirm that all "old" plantain cards have these keys?

mxcdoam, 2024-04-17:
@gornekich For all 10 cards I've tested - yes, they are same.

gornekich, 2024-04-18:
@mxcdoam could you try #3602 ? Does plantain parser reads all keys? Can you see correct parsed data?

mxcdoam, 2024-04-18:
while trying to checkout Git says:
'''
error: pathspec 'gornek/add_plantain_lrgacy_support' did not match any file(s) known to git
'''
But the update package from #3602 works great! Seems to be fixed :+1:

gornekich, 2024-04-19:
Fix arrived in dev. Thanks for testing

closed 2024-04-19 by gornekich