flipperdevices/flipperzero-firmware #3558: Access to Discord server

vanguacamolie, 2024-04-01:
Hello, I would like to be able to send messages in the Flipper Devices Discord server. Unfortunately, I need to give Discord access to my phone number in order to send messages there, which I am unwilling to do. Is there any other way to gain access? My username on Discord is `guacamolie`.

---

Sorry for the unconventional issue, but this seemed to be the best way to reach out since, well, I'm unable to send messages on Discord.

mxcdoam, 2024-04-01:
Second that, I want to participate, but can't if I don't give discord my phone number

skotopes, 2024-04-14:
@mxcdoam @vanguacamolie sadly that is how discord works, we can not do anything with it.

closed 2024-04-14 by skotopes