flipperdevices/flipperzero-firmware #3566: Mifare DESFire Emulate UID not working

Fabunator, 2024-04-03:
### Describe the bug.

I have a copy of my woking place badge from Version 0.97.x
In this version i could emulate the UID and use this to log into my work.
In version 0.99.1 this is not possible any more.
I also made a new copy with the current version but nothing changed.

The type of the Tag is ISO14443-3A (Mifare DESFire EV1 4k) in the nfc file the type is Mifare DESFire 

### Reproduction

Read NFC Mifare DESFire Tag
Emulate UID

gornekich, 2024-04-15:
Hello @Fabunator ! 
I can't reproduce any problems with UID emulation. We need more information here. The protocol below Mifare Desfire is ISO14443-4A, not 3-A. The UID emulation for iso-3a and iso-4a are different.

1. Is it possible that your reader starts to read more than just UID? Does the emulation still works on 0.97 version?
2. Do you have any Logs during emulation? If readers reads more than just UID, you will see Log button on the screen during emulation. After pressing OK, you will have a screen with raw data received from reader. If you have this, please share this raw command here

Fabunator, 2024-04-19:
I am sorry i made some mistakes in my description
The tag is of the Type ISO 14443-4 (Mifare DESFire EV1 4k)
[image]

I added 2 .nfc files from different Firmware Versions of the Flipper. (changed the UID for reasons)
The first one is from 0.93.0 the second from 0.99.1
In Version 0.93.0 the emulation of the UID works for both files and the reader can detect my badge
In Version 0.99.1 the reader is not detecting my emulated UID

```
Filetype: Flipper NFC device
Version: 3
# Nfc device type can be UID, Mifare Ultralight, Mifare Classic or ISO15693
Device type: Mifare DESFire
# UID is common for all formats
UID: 00 00 00 00 00 00 80
# ISO14443 specific fields
ATQA: 03 44
SAK: 20
# Mifare DESFire specific data
PICC Version: 04 01 01 01 00 18 05 04 01 01 01 04 18 05 04 50 53 CA E8 2E 80 BA 35 19 D5 10 49 12
PICC Free Memory: 3840
PICC Change Key ID: 00
PICC Config Changeable: true
PICC Free Create Delete: false
PICC Free Directory List: true
PICC Key Changeable: true
PICC Max Keys: 01
PICC Key 0 Version: 00
Application Count: 1
Application IDs: 00 21 F5
Application 0021f5 Change Key ID: 00
Application 0021f5 Config Changeable: true
Application 0021f5 Free Create Delete: false
Application 0021f5 Free Directory List: true
Application 0021f5 Key Changeable: true
Application 0021f5 Max Keys: 02
Application 0021f5 Key 0 Version: 00
Application 0021f5 Key 1 Version: 00
Application 0021f5 File IDs: 00 01 02 03
Application 0021f5 File 0 Type: 00
Application 0021f5 File 0 Communication Settings: 01
Application 0021f5 File 0 Access Rights: 00 10
Application 0021f5 File 0 Size: 32
Application 0021f5 File 1 Type: 04
Application 0021f5 File 1 Communication Settings: 00
Application 0021f5 File 1 Access Rights: 00 10
Application 0021f5 File 1 Size: 8
Application 0021f5 File 1 Max: 61
Application 0021f5 File 1 Cur: 0
Application 0021f5 File 2 Type: 00
Application 0021f5 File 2 Communication Settings: 01
Application 0021f5 File 2 Access Rights: 00 10
Application 0021f5 File 2 Size: 192
Application 0021f5 File 3 Type: 01
Application 0021f5 File 3 Communication Settings: 00
Application 0021f5 File 3 Access Rights: 00 E0
Application 0021f5 File 3 Size: 32
Application 0021f5 File 3: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

```
Filetype: Flipper NFC device
Version: 4
# Device type can be ISO14443-3A, ISO14443-3B, ISO14443-4A, ISO14443-4B, ISO15693-3, FeliCa, NTAG/Ultralight, Mifare Classic, Mifare DESFire, SLIX, ST25TB
Device type: Mifare DESFire
# UID is common for all formats
UID: 00 00 00 00 00 00 80
# ISO14443-3A specific data
ATQA: 03 44
SAK: 20
# ISO14443-4A specific data
T0: 75
TA(1): 77
TB(1): 81
TC(1): 02
T1...Tk: 80
# Mifare DESFire specific data
PICC Version: 04 01 01 01 00 18 05 04 01 01 01 04 18 05 04 50 53 CA E8 2E 80 BA 35 19 D5 10 49 12
PICC Free Memory: 3840
PICC Change Key ID: 00
PICC Config Changeable: true
PICC Free Create Delete: false
PICC Free Directory List: true
PICC Key Changeable: true
PICC Flags: 00
PICC Max Keys: 01
PICC Key 0 Version: 00
Application Count: 1
Application IDs: 00 21 F5
Application 0021f5 Change Key ID: 00
Application 0021f5 Config Changeable: true
Application 0021f5 Free Create Delete: false
Application 0021f5 Free Directory List: true
Application 0021f5 Key Changeable: true
Application 0021f5 Flags: 00
Application 0021f5 Max Keys: 02
Application 0021f5 Key 0 Version: 00
Application 0021f5 Key 1 Version: 00
Application 0021f5 File IDs: 00 01 02 03
Application 0021f5 File 0 Type: 00
Application 0021f5 File 0 Communication Settings: 01
Application 0021f5 File 0 Access Rights: 00 10
Application 0021f5 File 0 Size: 32
Application 0021f5 File 1 Type: 04
Application 0021f5 File 1 Communication Settings: 00
Application 0021f5 File 1 Access Rights: 00 10
Application 0021f5 File 1 Size: 8
Application 0021f5 File 1 Max: 61
Application 0021f5 File 1 Cur: 0
Application 0021f5 File 2 Type: 00
Application 0021f5 File 2 Communication Settings: 01
Application 0021f5 File 2 Access Rights: 00 10
Application 0021f5 File 2 Size: 192
Application 0021f5 File 3 Type: 01
Application 0021f5 File 3 Communication Settings: 00
Application 0021f5 File 3 Access Rights: 00 E0
Application 0021f5 File 3 Size: 32
Application 0021f5 File 3: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
I have also attached the logs while reading
[image]

Fabunator, 2024-04-19:
Back to open

Fabunator, 2024-05-16:
Are there some news on this case?
Any more Information i can provide here?

gornekich, 2024-05-16:
Thanks for information, it should be enough. Still working on this issue.

skotopes, 2024-07-07:
@gornekich any updates?

gornekich, 2024-07-18:
Still working on it

Fabunator, 2025-04-30:
Hi is this resolved with the improvements for DESFire in the 1.3.4 Release?

yakamoneye2018, 2025-11-10:
Addition: the emulation does not even react in the NFC tools app in Android.