flipperdevices/flipperzero-firmware #3575: GPIO Power Issue

Az-Neter, 2024-04-06:
### Describe the bug.

I was running a pi0 connected via 3v pin and ground with no uart.  
Power draw was ≈450mA, a % loss of around 5% every ten minutes.  
* After 20 minutes with this running, my battery level suddenly jumped, immediately from 58% down to only 7%.  
I disconnected the pi and recharged the flipper: it took two hours to go back to 99% battery level.  
The pi0 was idle on tty interface the entire time, with only stock processes running.  

### Reproduction

Don't fully discharge your flipper for six months and then consume more battery than chemically exists within the cell?  
I genuinely have no idea how this is possible.

### Target

Battery Health

### Anything else?

Health of battery should've at least altered to reflect this sudden dip. The battery stats screen otherwise made no indication why this happened.

skotopes, 2024-04-14:
Remaining battery capacity in flipper is calculated by bq27220 gauge. It is quite sophisticated on the inside and uses coulomb counter and voltage curves to precisely estimate remaining charge.

We can not say what exactly happened without detailed current/voltage/capacity log from the device.
Discharge and then charge device back, try to reproduce this issue, pay attention to the info from the device(can be obtained with mobile companion).

Az-Neter, 2024-04-14:
It's already been well over a week and I'm not going to remotely attempt to reproduce a glitch that will potentially damage my hardware. That's a ridiculous ask.

You may as well close this since your "sophisticated" gauge malfunctioning is apparently my fault, I guess.

Where would I find logs for this? Do you expect that I manually record the information for you, or would there be an output file on device that I can send you?

skotopes, 2024-04-15:
We've been designing device to be fool proof to some point.

However you really pushed it to the limit: RPI can easily draw more than 1.5A(even short burst is enough), which is current limits for charger/battery protection circuits(they will shutdown to prevent damage).

In this specific case there is nothing wrong with flipper. No need to attach anything.

closed 2024-04-15 by skotopes

Az-Neter, 2024-04-16:
I gave quite literally every detail possible as to why this happened and if you have the balls to claim this is my fault then you're more shortsighted than you've previously let on.
Trust me, I'm neither intelligent nor stupid enough to push this device to its limits.
20 minutes powering a pi over gpio is not pushing anything to any limits.
Was that actually a joke, or?

Either you're insane or you just don't care to fix when people bring you problems beyond your comprehension.
It should not have taken over a week for a maintainer to get to an open hardware issue.

No follow up on the log question I asked?

The team worked really hard to make this a robust system so it seems pretty arrogant that when a user brings you an obvious problem beyond your considerations that you'd call it their fault and ignore the issue.

I'm not the only person using a pi with flipper, and when multiple more devices start melting down since you refuse to even question this problem, I'll be here laughing at you in public.

I gave you 100% of the details I have. If that's still not good enough for you, that's kind of on you.

All I asked was that the battery health should've actually been able to recognize this was happening. That doesn't seem like an insane request beyond the scope of your development.
Just remember who actually keeps flipper going: your community. More of your pushes have come from cfw than from the official dev team.

Az-Neter, 2024-04-16:
I'm not asking you for a fix, I'm asking you to even remotely care that this happened.