flipperdevices/flipperzero-firmware #3582: SubGHZ: KeeLoq: transmitter manufacturer key

santomet, 2024-04-11:
### Description of the feature you're suggesting.

I have some [Satel brand transmitters](https://www.satel.pl/en/products/intruder-alarms/radio-controllers/keyfobs/t-4/) and I would like to try to get their manufacturer key.

From previous issues I know that @Skorpionm does the power analysis when he gets the transmitter physically. Of course, I would like to have his SW to use on the data I can obtain from an oscilloscope. But I can understand that you do not want to share your app.

However, I can not find any info about how to communicate with Skorpionm and where to actually send him the transmitter for analysis. Therefore, I am opening this issue and will wait for him to contact me :) 

Thanks in advance.

Skorpionm, 2024-04-14:
Good afternoon, well, first you need to look at what exactly is inside your keychains.  Can you take normal photos of the insides so that the markings of the parts can be seen?

santomet, 2024-04-15:
Good morning, thanks. Yes, I took one photo as only one side of the PCB has visible traces and some components:
[image]

Skorpionm, 2024-04-15:
In theory, it is possible to remove power curves, but in 80% of cases the manufacturer's key is unique for each key fob.  and the game is not worth the candle.  + we need delivery to Russia, but now it’s difficult

santomet, 2024-04-16:
Very well, according to the Satel manual, the learning process requires pressing any button on the fob twice. Also, there is explicit support for all Satel fobs. So there is apparently Normal learning only.

Therefore, I think that finding out Satel manufacturer's key is worth it in this case.

There are ways how to send anything to Russia but there might be alternatives. If possible, please contact me at [keeloq@santomet.eu](mailto:keeloq@santomet.eu) and we can talk about the details. Thanks!

Skorpionm, 2024-07-09:
It is not possible without this system in my hands

closed 2024-07-09 by Skorpionm