flipperdevices/flipperzero-firmware #3586: Js 'textbox' and 'math' inclusion

Az-Neter, 2024-04-15:
### Description of the feature you're suggesting.

There are .c modules written available for many more JavaScript features which are working and available. I do not own the repositories so I cannot make the pull requests.
Since the dialog function prevents the script continuing, the textbox module is a mandatory function to get the most out of the built in JavaScript port.
Please communicate with the devs of these modules to get them added to the ofw! I'm very excited to use the JavaScript features, yet there are still many limits which are addressed elsewhere waiting to be filled in.

### Anything else?

JavaScript cross scripting? It would be useful to invoke scripts from within other scripts, if possible.

Keep up the great work :)

Willy-JL, 2024-04-15:
I'm willing to contribute the custom modules we have made in Momentum if OFW is interested, I was planning on making a few pull requests soon when I had more free time regardless

skotopes, 2024-04-15:
@Willy-JL sure, bring it on

skotopes, 2024-07-07:
Done

closed 2024-07-07 by skotopes