flipperdevices/flipperzero-firmware #3591: Password protection for T55xx

mxcdoam, 2024-04-15:
### Description of the feature you're suggesting.

I'm talking about intercom keys. In my area it is a common countermeasure of service providers, to check if the tag is writeable. If intercom detects T55xx, it damages the tag (probably writing zeros as UID + locking or password protecting. That way tag cannot be detected and generally reused anymore). But it seems that tags password-protected by me are not detected as clones and continue to work. Right now flipper zero cannot protect tags, so I was wondering if it's possible to add this feature?

CountParadox, 2024-06-16:
This would be good

ThreeSixes, 2024-12-12:
+1 on this. I have tags that need to be password protected and it would be great if I could perform operations on a password protected device and be able to modify the password at will without having to carry my proxmark 3 around.

Astrrra, 2025-02-04:
I've got good news, this will be added in #4096

mxcdoam, 2025-02-05:
Thank you very much!

ThreeSixes, 2025-02-05:
I second this! I'm very excited to have that capability!