flipperdevices/flipperzero-firmware #3592: Ability to write MIFARE Ultralight tags

mxcdoam, 2024-04-15:
### Description of the feature you're suggesting.

This feature is pretty much as the title says: right now flipper zero can only read and emulate Ultralight tags. Is it possible to also implement writing within the FW? Are there any plans for that? I'm talking about simple Ultralight tags without passwords, counters and encryption.

mxcdoam, 2024-04-16:
> @mxcdoam full writing is implemented in Nfc Magic app(requires gen4 magic card). Normal cards can not be fully written because of the immutable blocks.

Sorry, I was talking about scenario: read card -> save -> edit data -> update original card, not touching UID and OTP. Also, MifareMagic doesn't support direct write ultralights, sadly.

skotopes, 2024-04-16:
@gornekich @Astrrra do we have any reason why we can not implement it? Like we have for MFC?

gornekich, 2024-04-16:
No reason. We already have write to initial card for NTAGs, it will dot be difficult to add this for Ultralights

closed 2024-06-19 by gornekich