flipperdevices/flipperzero-firmware #3599: Suggestion: Add NFC Tag Writing in the “More>” Menu (Emulate).

Killer74-hub, 2024-04-17:
### Description of the feature you're suggesting.

It would be nice to write info from an NFC Tag to another NFC Tag . Is that possible?

gornekich, 2024-04-17:
Hello @Killer74-hub 

Almost all cards have blocks / pages, which can't be modified. They are written by manufacturer and have read only access. That makes impossible to clone one card to another. The same is with wiping.

Meanwhile, there are specific "Magic" cards, which allows these blocks and pages modification. We have NFC Magic application in our catalog, which works with these cards. Using this app you can fully clone your Mifare Classic or Ultralight cards on Magic card. https://lab.flipper.net/apps/nfc_magic

mxcdoam, 2024-04-23:
Sounds like #3592 but for all tags supported. Not cloning, just writeable part writing/erasing