flipperdevices/flipperzero-firmware #3600: MIFARE DESFire Auth

einsjakob, 2024-04-17:
### Description of the feature you're suggesting.

[image]

Hi, as seen in the picture, I cant read the apps of my MIFARE DESFire card because it would need authentication. Is it possible to bypass or solve this in another way?

BR, Jakob

mxcdoam, 2024-04-17:
No, if this card is not issued by you and you don't have keys - it is not possible.

gornekich, 2024-04-17:
Hello @einsjakob 
Mf DESFire is secure protocol. Your card configured the way, you can't event read applications without authentication with master key. It's almost impossible to get master key if you don't know it.

We don't plan to support authentication with known key in nfc application, since it's very hard to develop convenient design for this task and few users know keys for their cards

closed 2024-04-17 by gornekich