flipperdevices/flipperzero-firmware #3618: FDI iPassan Mifare Plus emulation not working

Tomi2965, 2024-04-26:
### Describe the bug.

Emulation of original FDI Mifare Plus tokens on FDI iPassan reader not works. In log is only half of UID (see images). Token is marked as Mifare Classic 1k

[image]
Code readed on Flipper

[image]
Yellow line - emulated via FZ, green line - token readed by reader. 

Note: please see, that FZ readed UID in reverse order. 

Token model: FDI GB-010-230
Reader model: FD-020-086

If anybody need any beta-test, or any info, do not hesitate and just ask! I have one iPassan at home for testing (including destructive tests). 

### Reproduction

Emulation of original FDI Mifare Plus token on FDI reader

### Anything else?

Have a nice day! :)

gornekich, 2024-07-18:
Still working on it

Tomi2965, 2024-08-13:
@gornekich do you need any additional info?

Tomi2965, 2024-09-12:
@gornekich I forgot to mention, that emulation of NFC tokens on FDI iPassan readers causes a reader reboot after door open for unknown reason. This happens on various models of readers. Same happens on emulation of NFC for collect keys from reader.

Tomi2965, 2024-11-21:
@gornekich any progress?