flipperdevices/flipperzero-firmware #3630: Slix emulation still broken since NFC refactor

Didgitalpunk, 2024-05-02:
### Describe the bug.

Slix2 tag emulation still does not work after refactor.
tested against a Dymo labelwriter's DRM by scanning an authentic roll's tag then placing my flipper inside the printer where the tag would be.
Pre-refactor, works as intended, printer detect the emulated roll.
Post-refactor, same tag dump, no go. new tag dump, still no go.

tested with firmwares 0.90.1 for pre-refactor version, and 0.101.2 (since released, and a few before that as well)

### Reproduction

scan roll's tag.
emulate roll tag
place flipper inside the printer against the nfc reader
pre-refactor, printer detects the roll that's emulated, post-refactor, nothing happens.

gornekich, 2024-05-02:
@Didgitalpunk thanks for the report. Could you please specify exact model of your labelwriter and roll? We need it to reproduce the issue

Didgitalpunk, 2024-05-02:
printer's a labelwriter 550 turbo, tested with 28x89 standard adress labels as well as 36x89 Large aderss labels

mxcdoam, 2024-05-03:
I tried to emulate SLIX-S and Tag-It HF, both times - unsuccessful. The reader doesn't react and debug log says:
 [D][Iso15693_3Listener] Wrong CRC, buffer size: 8
and
[D][Iso15693Parser] Frame parse failed
[slix-s.log](https://github.com/flipperdevices/flipperzero-firmware/files/15197505/slix-s.log)

[tag-it.log](https://github.com/flipperdevices/flipperzero-firmware/files/15197507/tag-it.log)

Didgitalpunk, 2024-05-20:
Thanks for the potential fix! I'll try it out tomorrow, printer's at work :)

closed 2024-05-22 by gornekich

IIntuitiveNaterrey, 2025-07-22:
Hey @Didgitalpunk I hope you see this. would you be able to help me figure out what all steps you took to get this to work. I see a vague Idea of what you did, but I'm brand new to the flipper zero and NFC/RFID aside from setting up keys for offices.  
I'd like to figure out how to do this for a friend who runs a vet and they have to use dymo printers because of avimark.

IIntuitiveNaterrey, 2025-07-22:
Bummer, so no way to copy a new Dymo tag on to a fresh empty tag?