flipperdevices/flipperzero-firmware #3646: 125MHz Card Not Reading - Images and Raw Files Attached

davewilliamstx, 2024-05-13:
### Description of the feature you're suggesting.

Have tried two different versions of this card, not reading on Flipper. I'm not sure on source of the cards, our HOA uses them.

[Unknown RFID Card.zip](https://github.com/flipperdevices/flipperzero-firmware/files/15288222/Unknown.RFID.Card.zip)

[image]
[image]
[image]

skotopes, 2024-05-22:
Yep this one is RFID, payload ASK coded, should be totally doable.

zinongli, 2024-06-09:
The reading from .ask looks similar to #3336
`100000000011010010111111111111111111111011110011011100111011000011011010010111111111111111111111`
Reverting it and adding spacers gives Securakey 26-bit format
`preamble     ??bitlen   reserved        EPf   fffffffc   cccccccc   cccccccOP  CS1        CS2`
`0111111111 0 01011010 0 00000000 0 00000000 0 01000011 0 01000110 0 01001111 0 01001011 0 10000000 0 00000000 0 0000`
`f` deontes facility code and `c` denotes card id. In this case card id is `1010001100100111` which is equal to 41767 in decimal. The same card number as printed on OP's card. 

Also long pulses are around 330 mu sec given by Flipper Lab's pulse plotter. Close enough to Securaket's RF/40 clock which would give 320 mu sec. 

I'm not sure if reverting is intentional or is it my mistake in analyzing the signal. If it is intended to be like the second sequence, then my PR #3697 should enable support for this card as well. 

@davewilliamstx : do you mind taking a photo of the reader? It might help confirming the brand and protocol of this card.

davewilliamstx, 2024-06-12:
Sure thing, it’s pretty beat up - hope it helps.

zinongli, 2024-06-12:
Thanks! I'm not sure if the image got through from email replies to the comment section. Would you mind upload it on github directly?

davewilliamstx, 2024-06-12:
Ah, drat - here it is again. Cheers!
[image]

zinongli, 2024-06-12:
Ok I'm pretty sure that's a securakey radio key reader. Should be supported once my pull request above passes through, fingers crossed.

skotopes, 2024-07-07:
Fixed in https://github.com/flipperdevices/flipperzero-firmware/pull/3697 thanks to @zinongli

closed 2024-07-07 by skotopes