flipperdevices/flipperzero-firmware #3649: Add wait/pause timing for subghz protocol playback

smolbun, 2024-05-15:
### Description of the feature you're suggesting.

I came across a RF remote that uses princeton protocol,  I can only replay it using the save .sub file from "Read RAW".
Inspecting the signal out by replaying using the protocol directly shows ~7ms of pause/wait time before the next signal is played. In actuality the pause/wait time is ~9ms.

Would it be possible to measure the estimate of the pause/wait time and store it in the .sub similarly to how "TE" is measured?

URH image played directly using the protocol:
[image]

skotopes, 2024-05-22:
@Skorpionm what do you think?

smolbun, 2024-05-25:
Yeah its not receiving at all, no matter how long I press it. I was thinking if it could be dynamically calculated instead, like by taking the maximum or the average pause timing then dividing by te. Would that be a safer approach instead?

Skorpionm, 2024-05-26:
no, this will break the work of already saved Princeton files.  since they do not store the duration of the pause between sendings.  as an option, record the signal in Read raw mode with the threshold function enabled, only the useful signal will be recorded, almost without unnecessary noise

smolbun, 2024-05-26:
Ah yeah make sense, currently I'm recording at as raw then decoding it if I need to analyze the transmission. Would it then be possible to introduce an optional parameter so that it can fall back to the default 30 if not specified?

Skorpionm, 2024-05-26:
why are you decoding it?  if it’s a simple keychain, then the parcels don’t change; if you’re trying to force the system for issuing orders in food court, I’m against it.  Yes, and it’s wrong to drive in another crutch

smolbun, 2024-05-26:
I'm confused with your response, the captured transmission is from my fan remote. I'm looking for a more efficient way to store the decoded transmission with a custom pause timing without having to use the raw transmission.

Skorpionm, 2024-05-28:
check https://github.com/flipperdevices/flipperzero-firmware/pull/3671

smolbun, 2024-05-29:
It works, I got `Guard_time: 37`. Thanks for the implementation, you may close this now.

closed 2024-06-02 by Skorpionm