flipperdevices/flipperzero-firmware #3650: Incorrect subghz frequency bands for Thailand region (TH)

panangam, 2024-05-16:
### Describe the bug.

The subghz module does not allow transmitting frequencies in the range ~433Mhz, saying "Transmission is blocked". According to the [Thai website](https://www.nbtc.go.th/spectrum_management/%E0%B9%81%E0%B8%9C%E0%B8%99%E0%B9%81%E0%B8%A1%E0%B9%88%E0%B8%9A%E0%B8%97/%E0%B8%95%E0%B8%B2%E0%B8%A3%E0%B8%B2%E0%B8%87%E0%B8%81%E0%B8%B3%E0%B8%AB%E0%B8%99%E0%B8%94%E0%B8%84%E0%B8%A5%E0%B8%B7%E0%B9%88%E0%B8%99%E0%B8%84%E0%B8%A7%E0%B8%B2%E0%B8%A1%E0%B8%96%E0%B8%B5%E0%B9%88%E0%B9%81%E0%B8%AB%E0%B9%88%E0%B8%87%E0%B8%8A%E0%B8%B2%E0%B8%95%E0%B8%B4/National-Table-of-Frequency-Allocations-B-E-2555.aspx), this range is allowed for amateur radio. In fact, I live in Thailand, and have been receiving signals in this frequency range from various of my RF remotes already, so it seems to be widely used. 

[image]

[image]

### Reproduction

1. Update firmware to the latest version using the mobile app while in Thailand (not sure how the app figures out the region).
2. Navigate to Subghz -> Region Information and see that the ~433Mhz range is not listed. 

### Anything else?

It seems to be using the same frequency bands as the Japanese region. I would try to open a pull request, but somehow could not find anywhere in the source code where the "TH" region is even defined.

CookiePLMonster, 2024-05-16:
The region database indeed seems to assign Japanese frequencies to Thailand:
https://update.flipperzero.one/regions/api/v0/bundle
`"TH":["F_JP_312","F_JP_920"]`

panangam, 2024-05-17:
> The region database indeed seems to assign Japanese frequencies to Thailand:
> https://update.flipperzero.one/regions/api/v0/bundle
> `"TH":["F_JP_312","F_JP_920"]`

Interesting. Is there anyway to contribute to this list?

CookiePLMonster, 2024-05-17:
> Interesting. Is there anyway to contribute to this list?

Not that I am aware.

skotopes, 2024-05-21:
I've updated region data, should be correct now

closed 2024-05-21 by skotopes