flipperdevices/flipperzero-firmware #3668: NFC Ultralight C Unlock Manually Password Too Short

zinongli, 2024-05-27:
### Describe the bug.

When an Mifare Ultralight C card is read and found to be password protected, there is the option to Unlock - Enter Password Manually. ULC's password is 16-byte long, but when you enter this manually input interface, there are only 4 bytes allowed (see below).

[image]

There appears to be left and right arrows to the side of the password box, but pressing left or right button only allows one to select different bytes but not go to a next page, or a previous one. 

[image]

I've also got suggestions from NFC Gandalf and NFC Samwise on Discord to ask for a new feature where the default 3des key will be tried in the read procedure for ULC

### Reproduction

Read a password protected ULC card, go to Unlock - Enter Password Manually. You will discover that it won't allow going to the next 4 bytes.

mxcdoam, 2024-05-29:
Yes, it is a bit confusing. I wanted to unlock my tag with "BREAKMEIFYOUCAN!" key and saw only four bytes to fill in.

skotopes, 2024-06-02:
@gornekich should we give @Astrrra this task?

gornekich, 2024-06-02:
@skotopes yes. We need to add authentication for Ultralight C

miladhatami1393, 2024-06-03:
this task?

gornekich, 2024-06-19:
@zinongli could you please try #3720 ?

zinongli, 2024-06-20:
> @zinongli could you please try #3720 ?

Good stuff! But there has to be a little delay. I'm on a work trip and didn't bring the ULC card with me. I will once I get back home on 22nd.

zinongli, 2024-06-23:
@gornekich I can confirm I can type in the password and try unlocking the card. However, since I didn't know the password of the card, and my guessed password was probably wrong, it just stucked in the "Don't move" page. I'm sure you have ULC cards on your end of which you know the correct password.

There is a small caveat: when I typed in the wrong password in Unlock, and quit during "Don't move" page, try reading the card again without quitting the app, it was unable to read the card. It stucked at the "Don't move" page in Read. But after I quit the NFC app and reenter, Read works again.

closed 2024-09-28 by zinongli