flipperdevices/flipperzero-firmware #3669: Flipper should auto lock (when enabled) on turn on, and on use of 2FA key.

robot-penguin34, 2024-05-28:
### Describe the enhancement you're suggesting.

The flipper zero currently does not seem to lock after a power on, nor does it ask for a passcode when using 2FA keys. 

**Why is this important:**
The flipper is meant to be a device you bring everywhere with you, and as such malicious actors may be able to access your 2FA key, and using almost any other information about you (see oosint videos), log into whatever account you signed in with. Making this highly useful feature almost useless, maybe even a security risk.

Please consider this great enhancement

### Anything else?

Please also consider a feature to wipe the device after X failed passcode attempts.

**For the devs** who are constantly giving their time to work on this project:
 I am trying to learn C so I can contribute to requests, rather than add to the load. Thank you for your constant work.

robot-penguin34, 2024-05-28:
Devs, again thank you for your constant work.

Tohkie, 2024-05-29:
doesn't the flipper zero already natively support a pin code with the option to factory reset

robot-penguin34, 2024-05-30:
@Tohkie in response to your question:

On inspection going to `settings > desktop` (where you configure you pin, and other related content) there **does not** seem to be an option to reset the device with failed attempts. It does however mention that you will need to **_manually_** reset it if you forget your code. Unless there is a piece of code that puts your device in a secure state after enough attempts (like iOS), the pin is negligible.

skotopes, 2024-06-02:
To be honest Flipper is not a replacement for security keys and never will be.

Wiping device doesn't make much sense without full disk encryption for SD card.
There is a prototype of full disk encryption that I've made, but it will require sacrifices of performance and usability.

closed 2024-06-02 by skotopes

robot-penguin34, 2024-06-03:
No worries. Honestly, it would be considerably hard to figure out the account the key is paired with. Again, thanks for your work.