flipperdevices/flipperzero-firmware #3677: A JavaScript module for handling a button press

tlhunter, 2024-05-31:
### Description of the feature you're suggesting.

I would like to request an API for receiving a single button press. This could be very similar to the existing dialogue API except that it wouldn't clear the screen or display any messages. The developer would be expected to draw to the screen ahead of time.

The API could be very basic:

```js
let buttons = require("buttons");
let result = buttons.waitForButtonPress();
// result in "up", "down", "left", "right", "ok"
```

Ideally the back button could be caught as well but I understand that it is currently the only way to consistently exit a program and probably shouldn't be caught.

Ideally, assuming the code runs inside of a loop, if the user were to hold down a button continuously then it would just work as expected, triggering the button press with each iteration.

```js
while(true) {
let buttons = require("buttons");
let result = buttons.waitForButtonPress();
// result is read with each iteration even if button remains held
// of course, there is no guaranteed timing...
}
```

### Anything else?

I believe this would unlock a lot of potential games.

skotopes, 2024-06-03:
That is Input service bindings

portasynthinca3, 2024-08-28:
Currently working on it as part of the JS GUI module: #3841