flipperdevices/flipperzero-firmware #3678: a JavaScript API for loading and drawing from a spritesheet

tlhunter, 2024-05-31:
### Description of the feature you're suggesting.

I would like to request a JavaScript API for opening a bitmap format representing graphics and for drawing them to the screen. This is a powerful and efficient pattern for drawing graphics. 

Basically each sprite in a spritesheet is perhaps 8x8 pixels while the overall sheet is perhaps 64x64 pixels. Then one programmatically chooses one of the sprites in the sheet and provides a coordinate to draw the sprite at.

The API could look like this: 
```js
let sprites = require('spritesheet');
let sheet = sprites.load('mysheet.bmp', 8, 8); // w, h
sprites.drawToScreen(sheet, 3, 8, 16);
```
This draws the fourth entry (index 3 when 0 based) in the spritesheet to pixel coordinate x=8, y=16. The drawn entry is 8x8 pixels.

I used bmp as an example but it could be gif or pcx or whatever format Flipper already uses.

### Anything else?

This would allow for interesting game development.

skotopes, 2024-06-03:
We have XBM and various derivatives as a part of the Gui service. What you want is full fledged GUI system in JS.
We'll think about it.

portasynthinca3, 2024-08-28:
Currently working on it as part of the JS GUI module: #3841