flipperdevices/flipperzero-firmware #3687: 125kHz fob not read / Protocol implementation?

WilliamHF, 2024-06-05:
### Describe the enhancement you're suggesting.

Hi,

Flipper cannot read a T5577 fob used for gate access. 

The fob in itself is completely generic 
[image]

Attached are ASK/PSK raw reads. 
[nvqnq.zip](https://github.com/user-attachments/files/15587181/nvqnq.zip)

Any help in decoding is welcome and appreciated if doable!

I have some additional fobs I can raw read if necessary to have multiple reads here.. 

Thanks

skotopes, 2024-06-05:
Do you know anything about the gate control system itself? Vendor, model?

skotopes, 2024-06-05:
ASK looks good. Also if you have proxmark you can try to dump 5577 data.

WilliamHF, 2024-06-05:
I don't unfortunately. Anything else I could try? Can something be determined from the ASK coding? I can provide raw read from a different tag if it helps. 
Also, it seems the reader doesn't respond (no LED turning green or red) when emulating the raw ask data from the flipper.

skotopes, 2024-06-05:
Flipper RAW data was never meant to be emulated. As for protocol itself: someone need to reverse engineer it.

zinongli, 2024-06-23:
This is likely an Indala 224. It can be read with [this](https://github.com/flipperdevices/flipperzero-firmware/pull/3337#discussion_r1649580735)