flipperdevices/flipperzero-firmware #3689: Setting to ignore certain frequencies in subghz/frequency analyzer

X3r0C00L, 2024-06-05:
### Description of the feature you're suggesting.

I have in my city two frequencies that are always broadcasting, one in the 431mhz range and one in the 316mhz range. They are very strong signals.  I have the rabbit labs flux capacitor and when I am trying to look for any type of transmissions in my vicinity my flipper is overridden by the other stronger signals. 
Could we get a feature to add a way in frequency analyzer to add certain frequencies and possibly frequency ranges that would cause frequency anaylyzer to ignore those certain signals? This is making our fox hunting and geocaching games very difficult in our area.

skotopes, 2024-06-08:
Yep, there is such option, see `setting_user.example` in SubGHz assets folder.

closed 2024-06-08 by skotopes

X3r0C00L, 2024-07-10:
I was more thinking that an event listener could be put into the frequency analyzer to listen for the down button press since it is unused, which would trigger a new screen that would then list the frequencies and allow the end user to select ones to skip and then exit back to the frequency analyzer.

X3r0C00L, 2024-07-10:
As an example, 304.349mhz is constantly beating up the front end of my receiver when I turn down the RSSI to find weaker signals. Unfortunately that frequency is not in the list of standard frequencies. But this doesnt stop the receiver and the frequency analyzer app from getting hung on it. I believe that "and I dont know this for sure" but once the thread is active it has pulled a frequency range that it gets in either a list or array and iterates through that in a loop until a condition is met. I dont think that the origin frequencies that it loads up are the same as the standard frequencies that are hard coded into the source code. Another example My in my list of frequencies I have a small block that jumps from 345mhz to 348mhz but sometimes the frequency analyzer will get stuck on a strong signal on something like 346.1 which isnt even in my list of frequencies. Once the signal goes away and frequency anaylizer goes to add it to the list of found signals it chooses one from my list that is closest which would be 345. So this methodology doesnt work very well.