flipperdevices/flipperzero-firmware #3691: 418 is blocked in MX region but it's what a lot of doors use here

RnD-AIM, 2024-06-06:
### Describe the bug.

Most access control remotes use controls at 418 MHz. I had several that I was working with stored, and now I can no longer transmit at that frequency.

### Reproduction

[image]
Here's the image of one of the ones I have saved.

hedger, 2024-06-06:
If you believe the regional limitations for a particular country are incorrect, please provide links to official regulations that explicitly allow the frequency.

skotopes, 2024-06-08:
Mexico currently uses same bands as US and CA.
Please provide link to official band allocation table and some examples of the devices that operates in 418MHz band.

closed 2024-09-13 by skotopes