flipperdevices/flipperzero-firmware #3693: RFID 125khz: simple thin white card without markings

mrg00l, 2024-06-07:
### Description of the feature you're suggesting.

Can't read it.
[RFIDRAW240607.zip](https://github.com/user-attachments/files/15740191/RFIDRAW240607.zip)

mrg00l, 2024-06-07:
The reader (unknown) does not emit without a card, but when the card is presented, the RFID DETECTOR determines the frequency (125khz).

skotopes, 2024-06-08:
No meaningful data in payload. Most likely it's ReaderTalkFirst protocol or not LFRFID.
Do you have proxmark?

mrg00l, 2024-06-09:
I have a card. I have access with a reader (there are about a dozen of them). But so far I don’t understand how it works. Reader does not emit without a card, RFID DETECTOR show nothing. Only when card detected, it shows 125 khz.

skotopes, 2024-06-11:
Most likely it's a HiTag.

closed 2024-07-07 by skotopes