flipperdevices/flipperzero-firmware #3701: NFC 14443a I-Block chaining

bettse, 2024-06-10:
### Description of the feature you're suggesting.

Based on my read of iso14443_4_layer_decode_block (https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/nfc/helpers/iso14443_4_layer.c#L49-L63) I believe the flipper is lacking I-Block chaining support.  I've done a dump of a card interaction and the first is 0x12 (0b00010010) which I think is chaining based on this: https://github.com/RfidResearchGroup/proxmark3/blob/master/armsrc/iso14443a.c#L2953 .  I'm by no means a low level 14a dev, so I might be wrong.

I'd love to have I-block chaining so I could have further interaction with this javacard.

gornekich, 2024-07-02:
Working on this right now. Should be ready this week

gornekich, 2024-07-18:
Still working on it, should be ready on next release

bettse, 2024-10-16:
Someone linked me https://github.com/flipperdevices/flipperzero-firmware/pull/3918

Tested my ndef app with dev and 🎉  no iblock problem anymore!

closed 2024-10-16 by bettse