flipperdevices/flipperzero-firmware #3708: Flipper not detecting nfc card type

SpyGdev, 2024-06-12:
### Describe the bug.

When I try to read a mifare DESfire key fob (NFC), it is unable to recognize the card type. When I got the flipper i scanned my house key and it gave the information and the card type. Now after about a 1 month of use (do not think hardware damage). It is giving unknown info and just the UID. this can be conformed with the use TagInfo.
[image]
[image]
[image]

### Reproduction

Scan the MIFARE DESFire key fob with the Flipper.
Observe that only the UID is returned, and the card type is not recognized.

### Anything else?

Possible Causes:

Software bug or recent firmware update affecting NFC reading capabilities.
Potential degradation in NFC reading hardware, though hardware damage is unlikely

Other Key types
Sadly I do not have any other keys that i have previously saved and have with me

doomwastaken, 2024-06-14:
Tried with our office EV3 2k card as well as other available ones and they seem to be working fine. Do you have any other versions of this card available for test? 
Also if this card is available for sale, could you provide a link so we could order one. 

Tested our cards on latest dev / 0.102.3 release

doomwastaken, 2024-06-14:
Also try positioning card differently during read, moving the card in process might give this result

doomwastaken, 2024-06-17:
Greetings, @SpyGdev which firmware version did you use?

SpyGdev, 2024-06-19:
sorry for the late reply, I was using 0.98.2, but updating to latest has seem to fix the issue

closed 2024-07-07 by skotopes

ERoCx420, 2025-01-12:
im having an issue with this NFC also it reads it but when i emulate this type on my apt door it doesnt unlock. it just flashes  blue not green