flipperdevices/flipperzero-firmware #3725: Electra lfrfid protocol bug?

itrack, 2024-06-20:
### Describe the bug.

I tried to clone an Electra RFID Tag and couldn't do it, I get the following error:

[image]

If I try to rewrite the same tag it works (receive success message), if I try to write other tags it doesn't work anymore, I get the above error.

### Reproduction

1. save a tag
2. I try to write a new tag, after a while I get the error in the description

itrack, 2024-06-22:
In the meantime I tried different types of tags to try to write on.

I tried with the same firmware and tags as one person who succeeded the operation, in my case it didn't work. Is it possible a hardware difference?

xMasterX, 2024-06-22:
> In the meantime I tried different types of tags to try to write on.
> 
> I tried with the same firmware and tags as one person who succeeded the operation, in my case it didn't work. Is it possible a hardware difference?

Writing is possible only to T5577 tags that are not protected with password

itrack, 2024-06-22:
@xMasterX I tried on new empty tags, the same type he succeeded another flipper user.
In his case it worked, in my case it didn't.

itrack, 2024-06-22:
And I forgot to specify, if I write an Electra type RFID on a T5577 tag it doesn't work, if I write on T5577 any other tag (that is not Electra type) it works.
The problem seems to be only if the copied tag to be written is of the Electra type.

itrack, 2024-06-25:
I tried writing on 3 types of T5577 tags, only on one of them the writing is partial, also receiving an error message.

[image]

[image]

Astrrra, 2024-07-11:
@itrack can you please provide the files of the tags you're trying to write? The screenshots do not have the data needed to reproduce your issue

itrack, 2024-07-11:
Sure:

[electra.zip](https://github.com/user-attachments/files/16174386/electra.zip)

itrack, 2024-07-11:
and this one in rfid format: 
[33b.rfid.zip](https://github.com/user-attachments/files/16174469/33b.rfid.zip)

Astrrra, 2024-08-01:
@itrack I am unable to reproduce your issue with any of my T5577 tags, neither genuine nor clones.  Writing the file you provided to any of the T5577 tags and then reading it back results in an identical file to the one you sent. It seems that you either have a faulty/misconfigured card, or it isn't a T5577 card at all (maybe it's a clone that has less memory pages). 

Please use a different card, preferably from a different supplier than the one you're having problems with.

[image]
[image]

itrack, 2024-08-01:
@Astrrra could it be a problem with the device?
I tested 3 different tags, on all of them I could write anything else than Electra type data. 
It's a bit strange that I used exactly a tag that someone else tried to write Electra tag and everything worked ok for him, but it didn't work for me.
I use the latest version of the official firmware.

Thank you

Astrrra, 2024-08-01:
@itrack are all of those tags from the same shop and/or have the same form? If so, they may all have the same problem. 

It is also extremely unlikely that this is a problem with your specific Flipper Zero, since writing the first part of the data is successful. If it was a Flipper Zero issue, it either wouldn't work at all, or write garbage data to the tag, neither of which is happening here. Written data being cut off in the middle would be very unusual for a hardware issue.

If you have access to a Proxmark3 device, you can use it to check your tags and possibly find out what makes them different.

itrack, 2024-08-01:
@Astrrra The same tags bought from the same seller, I have 10 of the same type, I tested more because I thought there was a problem with one of them. I don't have access to a Proxmark3 device.

It's a strange situation and somewhat frustrating because I can't figure out exactly what it could be.  It's also strange that it only happens with this type of tags and not with others

Astrrra, 2024-08-01:
@itrack please try using tags from a different seller, this seems to be the easiest solution to your problem.

I'll close this issue because it isn't something we currently can provide further help with, please re-open it if that problem persists with tags from a different seller.

closed 2024-08-01 by Astrrra

itrack, 2024-08-01:
@Astrrra the tag that worked for another user was from the same seller but I have 2 other tags from completely different sellers, the issue is the same for all tags from all sellers.

Astrrra, 2024-08-01:
@itrack just to double-check, you have tags from different sellers, and a different Flipper Zero device has been able to write the data to it, but your device wasn't able to do this, right? This sounds very unusual, please double-check that with a second device, and if this is true - please contact http://support.flipper.net/ for help with your device. They'll be able to further verify that this is a hardware issue and give you guidance on your next steps.

itrack, 2024-08-01:
I recap :D
I have 3 types of tags, all of them can be written and read with any tag except the Electra type.
When I want to write electra, I get an error that it doesn't work, but it seems that one of these types of tags is only partially written.I bought this type of tag after talking to another user and he recommended me to buy it after he succeeded.
So on 2 types nothing is written, on one type of tag only partially. In the case of other types of cloned tag (Non electra), everything is written.

The other user tested with the same tag that I sent you and that you also managed to write.

mariusosta, 2024-08-03:
Hello @Astrrra , I have the same problem that @itrack describes with the Electra type rfid. I tried 5 different types of cards, on 3 of them the information is only partially written.

itrack, 2024-08-05:
Some logs from CLI with new tag:

```
78991 [D][LfRfidWorker] Read started
979560 [D][LfRfidWorker] Electra, 0, [03 E8 C3 0B 88 7E 1E AA]
979736 [D][LfRfidWorker] Electra, 1, [03 E8 C3 0B 88 7E 1E AA]
979831 [D][LfRfidWorker] Electra, 2, [03 E8 C3 0B 88 7E 1E AA]
979952 [D][LfRfidWorker] Read stopped
980472 [D][DolphinState] icounter 295, butthurt 0
986833 [D][GattChar] Updating Battery Level char
1005472 [D][LfRfidWorker] Data write
1005478 [D][LfRfidWorker] Start ASK
1005941 [D][LfRfidWorker] Read started
1007886 [D][LfRfidWorker] EM4100, 0, [00 03 55 14 F1]
1008016 [D][LfRfidWorker] EM4100, 1, [00 03 55 14 F1]
1008102 [D][LfRfidWorker] EM4100, 2, [00 03 55 14 F1]
1008191 [D][LfRfidWorker] Read stopped
1008245 [D][LfRfidWorker] Data write
1008247 [D][LfRfidWorker] Start ASK
1008709 [D][LfRfidWorker] Read started
1008881 [D][LfRfidWorker] Electra, 0, [03 E8 C3 0B 88 00 00 00]
1009020 [D][LfRfidWorker] Electra, 1, [03 E8 C3 0B 88 00 00 00]
1009167 [D][LfRfidWorker] Electra, 2, [03 E8 C3 0B 88 00 00 00]
1009294 [D][LfRfidWorker] Read stopped
1009348 [D][LfRfidWorker] Data write
1009350 [D][LfRfidWorker] Start ASK
1009812 [D][LfRfidWorker] Read started
1009984 [D][LfRfidWorker] Electra, 0, [03 E8 C3 0B 88 00 00 00]
1010123 [D][LfRfidWorker] Electra, 1, [03 E8 C3 0B 88 00 00 00]
1010261 [D][LfRfidWorker] Electra, 2, [03 E8 C3 0B 88 00 00 00]
1010397 [D][LfRfidWorker] Read stopped
1010451 [D][LfRfidWorker] Data write
1010453 [D][LfRfidWorker] Start ASK
1010476 [I][Dolphin] Flush stats
1010477 [I][SavedStruct] Saving "/int/.dolphin.state"
1010486 [D][StorageInt] Device erase: page 9, translated page: c1
1010492 [D][StorageInt] Device sync: skipping
1010494 [I][DolphinState] State saved
1010915 [D][LfRfidWorker] Read started
1011079 [D][LfRfidWorker] Electra, 0, [03 E8 C3 0B 88 00 00 00]
1011215 [D][LfRfidWorker] Electra, 1, [03 E8 C3 0B 88 00 00 00]
1011352 [D][LfRfidWorker] Electra, 2, [03 E8 C3 0B 88 00 00 00]
```

itrack, 2024-08-05:
for some context: "Electra, 0, **[03 E8 C3 0B 88 7E 1E AA]**" is the tag that has been read -> "EM4100, 0, **[00 03 55 14 F1]**" is the tag to be written ->"Electra, 0, **[03 E8 C3 0B 88 00 00 00]**" this is how it ends up after trying to be written