flipperdevices/flipperzero-firmware #3758: CC1101: Fixed-length packet mode not supported

krzys-h, 2024-07-06:
### Describe the enhancement you're suggesting.

The packet engine in CC1101 can be configured in two modes:
* Fixed packet length mode - where the packets are always `PKTLEN` long
* Variable packet length mode - where the packets are at most `PKTLEN` long, and the first byte of the packet has special meaning and determines the length

None of the builtin presets use the fixed-length mode, but it's obviously useful when trying to interface with a receiver that expects it, and can be configured using custom presets.

Currently, the `write_packet` and `read_packet` functions assume you are working with variable-length packets, and always inserts/removes the length byte for you. This makes it impossible to properly send/receive fixed-length packets.

The code is also written in a way that suggests that whoever wrote it didn't understand that the length byte is supposed to be a part of the packet, and not magic metadata of how much bytes are in the FIFO. Note how the CC1101 datasheet specifies that the byte is not automatically added/removed, just like the address byte, which we currently have no custom handling for.

[image]

Even more weirdly, the length byte handling is split between `read/write_packet` and `cc1101_read/write_fifo`. When writing, the `write_packet` function is responsible for adding the length byte, and when reading, the `cc1101_read_fifo` function uses it to determine how many bytes to read and strips it from the packet.

### Anything else?

I don't have any immediate use case for the fixed-length mode, I just thought it might be nice to fix along with #3755 since they touch the same part of code, and somebody will likely run into this problem in the future.

skotopes, 2024-07-07:
As you probably figured out we are not using modem(packet engine) part of cc1101.
It's quite limited in its abilities, got errata and overall make no sense for us.
So we are not planning to add what you want, but we'll accept PR if someone would like to implement it.

closed 2024-07-07 by skotopes

Skorpionm, 2024-07-07:
everything is possible. and transmission of packets of variable length and fixed. example of using batch mode
https://github.com/flipperdevices/flipperzero-firmware/blob/dev/lib/subghz/subghz_tx_rx_worker.c

directly for your case, you need to create a custom preset that meets your requirements and load it through
https://github.com/flipperdevices/flipperzero-firmware/blob/d39c3b377652bfcec954303b8caf058716d4ed46/lib/subghz/subghz_tx_rx_worker.c#L142

skotopes, 2024-07-07:
@Skorpionm looks like his case is more specific, see https://github.com/flipperdevices/flipperzero-firmware/issues/3755