flipperdevices/flipperzero-firmware #3761: A JavaScript module for GPIO

Himura2la, 2024-07-07:
### Description of the feature you're suggesting.

A JS module that provides an API for Flipper Zero GPIO should exist.
This would unlock a ton of signal processing/generation capabilities in the JavaScript interface, which is much easier and user-friendly than FAP platform.

### Anything else?

I have an idea to use Flipper Zero as an external source for Bastl Kastl, and make it a part of a modular synthesis setup, but I'm too lazy for writing and building FAPs 🥺
The complexity of environment required for FAP building, and the complexity of C API demotivates me.

portasynthinca3, 2024-08-19:
Current working on it: https://github.com/flipperdevices/flipperzero-firmware/pull/3841

portasynthinca3, 2024-10-15:
Added in #3841

closed 2024-10-15 by portasynthinca3