flipperdevices/flipperzero-firmware #3766: devboard_flash ARGS="-c dev" works with FBT, but not with uFBT

CookiePLMonster, 2024-07-08:
### Describe the bug.

As described in the title - `./fbt devboard_flash ARGS="-c dev"` flashes a `development` version of the firmware, while `ufbt devboard_flash ARGS="-c dev"`  stays with `release`. It would seem that with uFBT, `ARGS` are passed to `ufbt` and not `scripts/wifi_board.py`. See the logs below for more information.

According to `ufbt --help`, `ARGS=""` should work with uFBT too.

### Reproduction

1. Attempt to update the devboard via `./fbt devboard_flash ARGS="-c dev"`. Notice `[INFO] Using channel 'development'` in the logs.
2. Attempt to update the devboard via `ufbt devboard_flash ARGS="-c dev"`. Notice in the logs:
   ```
scons: *** Do not know how to make File target `dev' (H:\dev\flipper-zero\.ufbt\current\scripts\ufbt\dev).  Stop.
[INFO] Using channel 'release'
```

### Logs

```Text
./fbt devboard_flash ARGS="-c dev"
python scripts/wifi_board.py -c dev
2024-07-08 14:16:53,522 [INFO] Trying to find WiFi board using VID:PID
2024-07-08 14:16:53,525 [INFO] Downloading https://update.flipperzero.one/blackmagic-firmware/directory.json
2024-07-08 14:16:54,031 [INFO] Using channel 'development'
2024-07-08 14:16:54,031 [INFO] Using version '9447ee92'

...

ufbt devboard_flash ARGS="-c dev"
scons: Entering directory `H:\dev\flipper-zero\.ufbt\current\scripts\ufbt'

fbt: warning: App folder 'H:\dev\flipper-zero\flipperzero-firmware': missing manifest (application.fam)
LoadAppManifest, line 130, in file "H:\dev\flipper-zero\.ufbt\current\scripts\fbt_tools\fbt_apps.py"
python H:\dev\flipper-zero\.ufbt\current\scripts/wifi_board.py
scons: *** Do not know how to make File target `dev' (H:\dev\flipper-zero\.ufbt\current\scripts\ufbt\dev).  Stop.
2024-07-08 14:17:08,905 [INFO] Trying to find WiFi board using VID:PID
2024-07-08 14:17:08,908 [INFO] Downloading https://update.flipperzero.one/blackmagic-firmware/directory.json
2024-07-08 14:17:09,330 [INFO] Using channel 'release'
2024-07-08 14:17:09,331 [INFO] Using version '0.1.1'
```

hedger, 2024-07-08:
There are 2 sides of the problem here.
First, ufbt was lacking "ARGS" commandline parameter definition. Easily fixable.
However, after introducing that parameter, there's a second issue, caused by hacky way of environment construction in ufbt. I'm looking for better solutions than the current one - and any of them will include a new release to ufbt Python package.

hedger, 2024-07-09:
@CookiePLMonster 

Could you please check if recent fixes solve the issue for you?

```shell
python3 -m pip install --upgrade --pre ufbt
ufbt update -b hedger/ufbt-args
ufbt devboard_flash ARGS="-c dev"
```

CookiePLMonster, 2024-07-09:
> ```shell
> ufbt devboard_flash ARGS="-c dev"
> ```

The issue seems resolved:

```
ufbt devboard_flash ARGS="-c dev"
scons: Entering directory `H:\dev\flipper-zero\.ufbt\current\scripts\ufbt'

fbt: warning: App folder 'C:\Users\Adrian': missing manifest (application.fam)
LoadAppManifest, line 130, in file "H:\dev\flipper-zero\.ufbt\current\scripts\fbt_tools\fbt_apps.py"
python H:\dev\flipper-zero\.ufbt\current\scripts/wifi_board.py -c dev
2024-07-09 17:54:05,011 [INFO] Trying to find WiFi board using VID:PID
2024-07-09 17:54:05,014 [INFO] Downloading https://update.flipperzero.one/blackmagic-firmware/directory.json
2024-07-09 17:54:05,175 [INFO] Using channel 'development'
2024-07-09 17:54:05,176 [INFO] Using version '9447ee92'
```

skotopes, 2024-07-09:
Nice ;-)

closed 2024-07-09 by skotopes

CookiePLMonster, 2024-07-09:
Shouldn't the issue be opened until the PRs are merged though?

PR #3770 (2024-07-10): ufbt: added ARGS to commandline parser
# What's new

- To be finalised when command-line forwarding from ufbt wrapper is fixed
- Fixes #3766

# Verification 

- Requires pre-release version of ufbt package
- https://github.com/flipperdevices/flipperzero-firmware/issues/3766#issuecomment-2217884092

# Checklist (For Reviewer)

- [x] PR has description of feature/bug or link to Confluence/Jira task
- [x] Description contains actions to verify feature/bugfix
- [x] I've built this code, uploaded it to the device and verified feature/bugfix