flipperdevices/flipperzero-firmware #3772: MIFARE Ultralight C hex dump shows bogus data for locked pages

supersat, 2024-07-09:
### Describe the bug.

At a recent event, we gave everyone MIFARE Ultralight C wristbands with some pages locked as part of a CTF. Many people tried reading their wristband with their Flipper Zero, and unfortunately, rather than seeing some pages locked in the hex dump, they saw bogus data (seemingly copied starting from page 0). The NXP TagInfo app for Android correctly showed those pages as XX XX XX XX.

### Reproduction

1. Auth-protect some pages on a MIFARE Ultralight C card. This can be done by writing 25 00 00 00 to page 0x2a and 00 00 00 00 to page 0x2b. This locks pages 0x25 and up from being read without authentication.
2. Read the tag with the Flipper Zero.
3. Select Info, then more, then scroll down to the bottom. The last 3 pages should show as locked, but are copies of pages 0, 1, and 2.

### Anything else?

FW version 0.103.1

gornekich, 2024-07-12:
Thanks for the report, working on it

mishamyte, 2025-04-22:
Hey @supersat,

Sorry, checked that repo for some UL-C related tickets and found yours.
It looks very similar to #3945 , which was fixed by #3959 .

Can you pls validate and confirm if that issue is still relevant?

Many thanks!