flipperdevices/flipperzero-firmware #3773: U2F not working at all

brneor, 2024-07-09:
### Describe the bug.

When I access the "U2F" app, I'm briefly presented with the message "Connect to a device" and then "Connected!", but after a couple of seconds it goes back to "Connect to a device". If I do the same but it's not connect to any USB device, it keeps saying "Connected".

I'm on latest stable firmware, 0.103.1. Tried to reboot, reinstall firmware and even a custom firmware, same issue.

### Reproduction

1. Just open the U2F app.

doomwastaken, 2024-07-09:
Greetings. What firmware version are you using and what OS? 

Just tried Mac/Win11 on latest release / latest dev and see no issue. In a bit will try on Ubuntu

brneor, 2024-07-09:
@doomwastaken I have to apologize. Even though the Flipper is saying "_Connect to a device_", **it is** working properly when I try to use it (tested with [this website](https://demo.yubico.com/webauthn-technical/).

Is the message "_Connect to a device_" an expected behavior in this scenario? I tested in a Windows 11 and a macOS 14 machine, both showed the same behavior.

doomwastaken, 2024-07-10:
Trying same thing on windows at this second. 

U2F open, USB cable not plugged in. Site is open and on "Insert or tap your YubiKey"
Separate window promt appears to connect external device to USB key, clicking ok
Plugging USB cable
Flipper says "Press OK to register"
Clicking ok then leads yubico registration complete and u2f app asks to be connected to device (USB cable plugged in)
Clicking authenticate on website instantly makes flipper go into "Press ok to authenticate"
Unplugging cable here works fine, re-plugging cable sends it to connected -> "Press ok to authenticate" on flipper.

Is that same behavior for you? Only part that I am not certain is behavior after registration, maybe app should state that it is connected, rather than "connect me to a device". Other than that, app works just fine on Mac/Ubuntu/Win10/Win11 that I had available for tests. 

Once I check the behavior of an app according to our design guidelines, I will get back to the ticket

brneor, 2024-07-10:
Yep, the behavior is _almost_  the same, except that my Flipper says "_Connect to a device_" all the time even if it's connected and sometimes says "_Connected_" when not connected. That was the culprit of my confusion.

doomwastaken, 2024-07-10:
I've encountered that behavior once, in firmware 0.92 or so, however device reboot fixed it and I was never able to reproduce it. Either way, I had opened a ticket, we will get back to you once it is merged into dev

brneor, 2024-07-10:
Ty for your attention!

doomwastaken, 2024-09-05:
Could you please try to reproduce the issue on latest firmware?

brneor, 2024-09-06:
Can't reproduce in the latest firmware :)

closed 2024-09-06 by doomwastaken