flipperdevices/flipperzero-firmware #3780: Wrong allowed frequency for SM

ctosae, 2024-07-10:
### Describe the bug.

San Marino (state within Italy) adopts the same rules as Italy and the European community as permitted frequencies, but not correctly mapped in Flipper

### Reproduction

Geolocate in San Marino

hedger, 2024-07-11:
Please provide exact discrepancies between current limitations and what's legally allowed. Also please back it up with government-issued spectrum regulations stating permitted frequency ranges.

ctosae, 2024-07-12:
I'm referring in particular to 433MHz (remote controls, etc.).
San Marino is a very small state within Italy (where the Italian language is spoken) and there are no these documents, but there are gates and automations at 433MHz as in all of Italy and Europe

hedger, 2024-07-12:
Thank you, a fix is coming soon.

skotopes, 2024-07-22:
Fixed, please update firmware

closed 2024-07-22 by skotopes