flipperdevices/flipperzero-firmware #3792: chubb 125khz tag support

austinscreations, 2024-07-16:
### Description of the feature you're suggesting.

To extend existing 125khz rfid tag support:

I'm working on a project that uses a rather old 125khz rfid tag - chubb branded 125khz card has label "G-prox II" I've only found one other company with an "off the shelf" reader that works in my application but has been very painful process to buy was hoping the flipper would help be design my own reader.

I have roughly 20-25 cards on hand that I can provide RAW data from. Plus have two different readers that can read these cards.
I'm not a good c programmer and currently busy in school so not sure I can offer much in the way of programming (also in no rush) but happy to help support this project (and quite happy with my new to me flipper)

here is an initial scan I did of one of the cards, and I'll await direction from someone on if the protocol is possible and what / if anything I can send to help.
[raw testing (card 054495).zip](https://github.com/user-attachments/files/16256651/raw.testing.card.054495.zip)

thanks

skotopes, 2024-07-23:
Looks like dump to contains payload, we'll implement it later.

austinscreations, 2024-08-09:
I loaded [741329a](https://github.com/flipperdevices/flipperzero-firmware/commit/741329a74351ffa9ac0e12f03aec1d42845f741e) onto my flipper via qflipper
and PR Add lfrfid GProxII support #3796 does solve the read on my specific card. (my cards happen to be 36bit cards)

thanks for the implementation.

skotopes, 2024-08-09:
Awesome, I'll close this issue then.

closed 2024-08-09 by skotopes